FreeCAD/FreeCAD #17223: TechDraw performance with medium sized assemblies is incredibly poor.

Herrcookiekiller, 2024-10-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When inserting larger projects as view into a TechDraw Page, the performance when viewing said page is incredibly slow and laggy.
A threshold as to when it starts lagging has not been determined yet.

```
My PC Specs are:
RTX4080
Ryzen 7 5800X
24GB RAM
```
Here is a video showcasing the poor performance compared to the 3D Viewport.
3D-Viewport = ~144fps
TechDraw Page = around 5fps ?

https://github.com/user-attachments/assets/1b2ac5e2-f582-4883-b6ec-c0c21bd0ec9f

I have already discussed this problem a bit on the Discord and enabled OpenGL support, this made the TechDraw performance jump up to around 15fps (guessed) but the 3D-Viewport became much more unresponsive and laggy.

Trying out different Render-Cache and Anti-Aliasing settings also dont appear to be altering performance.
(just tried it out anyways, even if it's just settings for the 3D-Viewport)

Two test files with TechDraw-ings are attached.
[example-files.zip](https://github.com/user-attachments/files/17355744/example-files.zip)

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus 0.4.62
  * Assembly3 0.12.2
  * Assembly4 0.50.12
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.0
  * freecad.gears 1.2.0
  * InventorLoader 1.4.0
  * kicadStepUpMod 11.1.0
  * lattice2 1.0.0
  * Lithophane
  * Manipulator 1.5.7
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.3.16
[/code]
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-10-13:
Nothing to do with OpenGL or other video settings.

There is a design flaw in TD in that it redraws the object many more times than necessary.

This isn't a complicated fix, but it will touch a lot of code.

luzpaz, 2024-10-13:
@WandererFan is there an open ticket for that issue ?

luzpaz, 2024-10-13:
@Herrcookiekiller JFYI, we are on the verge of releasing FreeCAD v1.0 and we are currently in the 2nd Release Candidate phase. Please consider migrating to RC2 (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2)

WandererFan, 2024-10-13:
There is now: #17226

luzpaz, 2024-10-13:
@WandererFan should we close this as a duplicate then ?

WandererFan, 2024-10-13:
I would close this one.  It might send a dev down a turkey trail looking for video changes.

luzpaz, 2024-10-14:
Closing in favor of #17226

closed 2024-10-14 by luzpaz