FreeCAD/FreeCAD #17224: Cannot let go of toolbars when dragging them

zorael, 2024-10-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If I click and drag a toolbar, I become locked in the drag operation. I cannot let go of the lifted toolbar and it will follow the mouse cursor until I exit. I can still use the program, but as soon as I move the mouse to an empty toolbar space, whatever toolbar I lifted will move there (but will never be placed).

Clicking to let go of the toolbar to place it does nothing. ESC doesn't do anything either. Even if I right-click the toolbar area and lock the toolbars, the lifted toolbar will continue to track the cursor, even without a grab handle.

### Full version info

```shell
OS: EndeavourOS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: makepkg
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * A2plus 0.4.67
  * Curves 0.6.47
  * fasteners 0.5.29
```

### Subproject(s) affected?

None

### Anything else?

https://github.com/user-attachments/assets/3a1230f9-90de-4e72-b8a9-ed0cbfb6b682

I am running kwin on Wayland.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-10-13:
I'm guessing this is a Wayland specific issue as there's no problems using this self-compiled build on X11:

```
OS: EndeavourOS (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git)
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.12.4, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * FreeCAD-themes 2024.7.24

```

benj5378, 2024-10-13:
@luzpaz can you please assign me, such that I don't forget

benj5378, 2024-10-13:
I cannot reproduce on my Wayland device. Has this been an issue on previous releases, @zorael? Do you use mouse, touchpad or anything else? It shouldn't be, that you have an mouse click counter installed or something (just brain storming)?

thothloki, 2024-10-14:
@zorael I tried on endeavour os with Wayland with the 1.1 appkmage and did not have that issue. 
Are you using a touchpad? I have had a similar issue with my HP Envy touchpad when my laptop was overworked (not with FreeCAD but in general).

zorael, 2024-10-15:
@benj5378
@thothloki

I am using a touchpad yes, but the behaviour remains if I use a bluetooth mouse. I am not using a mouse click counter or anything similar (would such a thing even work on Wayland?). I am using a desktop scaling of 150%, if that matters.

If I delete my freecad configuration files (`~/.config/FreeCAD`) letting go of toolbars seems to be working as it should again, so something in my settings must be triggering this problem. I have not tried to rebuild my configuration from scratch to see what change causes it.

I'm attaching a zip of the files.

[FreeCAD.zip](https://github.com/user-attachments/files/17376894/FreeCAD.zip)

luzpaz, 2024-10-15:
> **If I delete my freecad configuration files (~/.config/FreeCAD) letting go of toolbars seems to be working as it should again**, so something in my settings must be triggering this problem. I have not tried to rebuild my configuration from scratch to see what change causes it.

Yet another usecase for:  
* #16370

zorael, 2024-10-15:
I reset the configuration and immediately ran into this again, so I did some basic troubleshooting.

With virgin settings, the theme is FreeCAD Classic, and dragging and dropping toolbars works as expected.

If I change the theme to either FreeCAD Dark or FreeCAD Light, dropping immediately breaks. If I change back to FreeCAD Classic, it still stays broken until I restart the program, but after a restart it works as it should again.

FreeCAD Classic isn't very pleasant with dark colours, though. :c

karliss, 2025-02-10:
I can repeat the bug even settings erased and FreeCad started in safemode. To make sure it's not an upstream Qt toolbar bug, I tried a qt sample project which has toolbars and that one was working without issues.

muezabdalla, 2025-07-24:
I was facing a similar issue but it was only on labwc I could not reproduce it on wayfire so I reported it to labwc
https://github.com/labwc/labwc/issues/2931

@zorael  can you try again on a modern wayland compositor (e.g wayfire) and with modern freecad build or with the weekly build

zorael, 2025-11-25:
I cannot reproduce it on my laptop anymore, using the 1.1rc1 AppImage, but I have changed distro since I originally filed the issue.

```
OS: Aurora (Version: latest-43.20251124.1) (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0rc1.20251125 (Git shallow) AppImage
Build date: 2025/11/25 17:53:08
Build type: Release
Branch: (HEAD detached at 5376760)
Hash: 53767601ea65de7cd9f6590e7d5856afc0841635
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Object center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/161.377/1.5
```

Tested on a desktop machine running the original EndeavourOS/Arch and could not reproduce it there either.

Perhaps @karliss has better luck?

(It's hard to prove a negative.)