FreeCAD/FreeCAD #17226: TechDraw - Too many unnecessary redraws.

WandererFan, 2024-10-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There is a design flaw in TechDraw in that it redraws objects on each property change.   This is how TD has worked from its inception.

Many of these redraws could be eliminated by using some of the abilities of the QGraphicsScene - ex changing the graphic's position in the scene instead of redrawing at the new position.

### Full version info

```shell
any version since TechDraw WB was added.

OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 +1 (Git)
Build type: debug
Branch: fixDimAutocorrect17189
Hash: fcd60659ab43e9369a4be46e086b29e162571266
Python 3.10.12, Qt 5.15.2, Coin 4.0.0, Vtk , OCC 7.8.0
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.21
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-17:
@WandererFan is this something you're working on or would you be open for help with this ?

luzpaz, 2024-10-19:
I'm curious to see how much faster this will be once it's optimized.

WandererFan, 2024-10-20:
I'm always open to help.

In the interest of expectation management, this will help with scrolling/panning/zooming responsiveness,  but the big performance impact will still be the OCC routines which is another problem.

luzpaz, 2024-12-15:
> Many of these redraws could be eliminated by using some of the abilities of the QGraphicsScene - ex changing the graphic's position in the scene instead of redrawing at the new position.

@WandererFan are there any examples of this method being utilized in main already ?

FEA-eng, 2025-08-20:
I guess it's still relevant.

luzpaz, 2025-08-20:
@ryankembrey any chance you can take a look at this ?

ryankembrey, 2025-08-20:
@luzpaz I appreciate the thought, but I'm unfortunately too busy at the moment to contribute anything substantial to FreeCAD mainline. I'm focussing on the DFM workbench (my capstone project), and all of my other university classes at the moment. 

It is something I'm interested in, though. Just a bit short on time atm, and trying to manage priorities :)

FEA-eng, 2025-10-20:
Appears to still be relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

luzpaz, 2025-12-16:
@owfischer can you please test the 1.0RC1 and the weekly development versions? (https://github.com/FreeCAD/FreeCAD/releases)

Also please always add your full [About](http://wiki.freecad.org/About) info when reporting results from tests. TIA

owfischer, 2025-12-16:
> [@owfischer](https://github.com/owfischer) can you please test the 1.0RC1 and the weekly development versions? (https://github.com/FreeCAD/FreeCAD/releases)
> 
> Also please always add your full [About](http://wiki.freecad.org/About) info when reporting results from tests. TIA

@luzpaz  Sorry, I should have done that!

Here is a little test I did. I closed all other applications except FreeCAD and TaskManager and opened a specific FCStd file containing drawings. I took the time from initiating the respective action until the CPU load went back to 0%. The durations are reproducible by roughly +/-10%.

I guess, it's quite interesting for the developers who know what they've done:

## Results
### 1.0.0 RC1
- open file: 0:48 min
- pan drawing: 0.2 s
- recompute drawing: 0:25 min

### 1.0.0
- open file:  5:19 min
- pan drawing: 6 s - 17 s
- recompute drawing: 3:08 min
==>This version was really a pain!

### 1.1.0dev weekly build 20251031
- open file: 1:47 min
- pan drawing: 0.2 s
- recompute drawing: 0:48 min

### 1.2.0dev weekly build 20251210
- open file: 1:55 min
- pan drawing: 0.2 s
- recompute drawing: 0:14 min

## "About" info for the used versions:
### 1.0.0 RC1
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods:
- 3D_Printing_Tools
- A2plus 0.4.60e
- Assembly4 0.12.4
- fasteners 0.5.37
- Help 1.0.3
- Manipulator 1.4.9
```

### 1.0.0
```
OS: Windows 8 build 9200
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods:
- 3D_Printing_Tools
- A2plus 0.4.60e
- Assembly4 0.12.4
- fasteners 0.5.37
- Help 1.0.3
- Manipulator 1.4.9
```

### 1.1.0dev weekly build 20251031
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.20251031 (Git shallow)
Build date: 2025/10/31 03:23:50
Build type: Release
Branch: main
Hash: c270a8ac666dff22a92baf2a3a4f5200197bb678
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/87.3608/1.25
```

### 1.2.0dev weekly build 20251210
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20251209 (Git shallow)
Build date: 2025/12/09 13:53:17
Build type: Release
Branch: grafted,grafted
Hash: 5ad6a1ba58c665c23f1d217c23e74379d8c1a5a4
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/87.3608/1.25
```

Hope this helps!