FreeCAD/FreeCAD #17228: Part Design: Pad To first and Up to shape issues in a model with multiple solids within a single Body

FEA-eng, 2024-10-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This video shows the problem: https://youtu.be/HIVWPF8ieMI?si=5CyqED_qnVx6HK0T&t=1479

With "To first", pad doesn't stop where it should and extends way too far:

[image]

With "Up to shape" it seems to work but there's a strange artifact in the top part (where the pad starts):

[image]

[To first too far.FCStd.zip](https://github.com/user-attachments/files/17356296/To.first.too.far.FCStd.zip)

Experimental support for multiple solids within a single Body must be enabled. 

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 (Git)
Build type: Release
Branch: main
Hash: e478192aa5393de9b7df190341a043f79ec6ac49
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-10-13:
Confirmed, it works only if you select the intersected faces, no less no more.
[image]

FEA-eng, 2024-10-13:
> Confirmed, it works only if you select the intersected faces, no less no more.

Thanks. Potentially fixable?

> doesn't overlap the profile

Do you mean that that the sketch can't extend beyond the first encountered face for this mode to work?

Also, how did you visualize it this way?

FlachyJoe, 2024-10-13:
> Potentially fixable?

I'd like to detect faces intersected backward and remove them from the shape used to delimit the prism. It could help the algorithm.

> the sketch can't extend beyond the first encountered face for this mode to work

Yes, in case of planar face it's always ok because they are used as infinite plane but for the faces which are closed when extended it fails as you show here.

> how did you visualize it this way

I've created the torus from a circular sketch based on your. :stuck_out_tongue_winking_eye:

FEA-eng, 2024-10-13:
> I'd like to detect faces intersected backward and remove them from the shape used to delimit the prism. It could help the algorithm.

Would be nice if it worked this way.

> Yes, in case of planar face it's always ok because they are used as infinite plane but for the faces which are closed when extended it fails as you show here.

I guess that’s the case with concave faves mentioned in the documentation. For convex ones it works or all curved faces are problematic?
 
> I've created the torus from a circular sketch based on your. 😜

It looks just like a preview of subtractive operations from Link Branch so I thought you used something like that ;-)

FlachyJoe, 2024-10-22:
> I guess that’s the case with concave faves mentioned in the documentation. For convex ones it works or all curved faces are problematic?

If the projection of the sketch center of gravity doesn't intersect the extend face then the algorithm don't see it.
For flat face it only fails if face is parallel to the extrusion direction but for curved faces its more frequent:

[image]