FreeCAD/FreeCAD #17230: Button does not show when it is selected when using additive pipe

EvEggelen, 2024-10-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When adding edge when using the additive pipe, the "Add Edge" button is normally pressed. Then the button changes color ( darker) and the edge can be selected. This is the cased in the "No style sheet" is selected. When FreeCAD dark or FreeCAD light is selected, the color does not change anymore. When you are a new user ( like me ), this is confusing.

For quick testing I added the cad file I used.

[image]
(Button "Add Edge" is grey)
[image]
(No color change)
[image]

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

### Subproject(s) affected?

None

### Anything else?

[additivePipe.zip](https://github.com/user-attachments/files/17356523/additivePipe.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-10-13:
I'm guessing the two stylesheets are missing:

```
QSint--ActionGroup QFrame[class="content"] QToolButton:checked {
    color: white;
    background-color: qlineargradient(spread:pad, x1:0, y1:0.3, x2:0, y2:1, stop:0 #406350, stop:1 #68917a);
}
```
[image]

(note the colors won't suit either style but can be seen to function) pinging @MisterMakerNL

MisterMakerNL, 2024-10-13:
This is a good catch! And @Syres916 thanks for the fix!
Now I had a bit of a internal discussion if I should stick to classic style and just make the button look idented/pressed.
And I really think this isn't obvious enough, so I used the focus color too add a border: (since the button is in focus since it is active I though it was more suitable than the select color.)
[image]
[image]

closed 2024-10-14 by chennes