FreeCAD/FreeCAD #17232: Toponaming: duplicate element mapping errors in PD when creating Pad from partial sketch

hyarion, 2024-10-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a pad feature in part design I get duplicate element mapping errors.

Steps to reproduce:
1. Create two rectangles in the same sketch (or load the attached file: [testcase.FCStd.zip](https://github.com/user-attachments/files/17356754/testcase.FCStd.zip))
2. Select the edges of the small rectangle
3. In Part Design, click the Pad command tool button

The Pad is created but a bunch of error messages are displayed.

Screen cast for steps above (starts after step 1 is completed)

https://github.com/user-attachments/assets/1b171e92-7fd1-4e5f-8a65-cd257ecd5df4

<details><summary>Error messages</summary>
<p>

```
20:51:09  <ElementMap> <string>(1): duplicate element mapping 'g7v2;SKT;:H122e,V -> g7v2;SKT;:H122e,V;D1;:H122e:3,V Vertex3/Vertex2
20:51:09  <ElementMap> <string>(1): duplicate element mapping 'g5v1;SKT;:H122e,V -> g5v1;SKT;:H122e,V;D1;:H122e:3,V Vertex5/Vertex4
20:51:09  <ElementMap> <string>(1): duplicate element mapping 'g5v2;SKT;:H122e,V -> g5v2;SKT;:H122e,V;D1;:H122e:3,V Vertex7/Vertex6
20:51:09  <ElementMap> <string>(1): duplicate element mapping 'g6v2;SKT;:H122e,V -> g6v2;SKT;:H122e,V;D1;:H122e:3,V Vertex8/Vertex1
20:51:09  <ElementMap> <string>(1): duplicate element mapping 'g7v2;SKT;:H122e,V -> g7v2;SKT;:H122e,V;D1;:H122e:3,V Vertex3/Vertex2
20:51:09  <ElementMap> <string>(1): duplicate element mapping 'g5v1;SKT;:H122e,V -> g5v1;SKT;:H122e,V;D1;:H122e:3,V Vertex5/Vertex4
20:51:09  <ElementMap> <string>(1): duplicate element mapping 'g5v2;SKT;:H122e,V -> g5v2;SKT;:H122e,V;D1;:H122e:3,V Vertex7/Vertex6
20:51:09  <ElementMap> <string>(1): duplicate element mapping 'g6v2;SKT;:H122e,V -> g6v2;SKT;:H122e,V;D1;:H122e:3,V Vertex8/Vertex1
20:51:09  <ElementMap> <string>(1): duplicate element mapping 'g7v2;SKT;:H122e,V -> g7v2;SKT;:H122e,V;D1;:H122e:3,V Vertex3/Vertex2
20:51:09  <ElementMap> <string>(1): duplicate element mapping 'g5v1;SKT;:H122e,V -> g5v1;SKT;:H122e,V;D1;:H122e:3,V Vertex5/Vertex4
20:51:09  <ElementMap> <string>(1): duplicate element mapping 'g5v2;SKT;:H122e,V -> g5v2;SKT;:H122e,V;D1;:H122e:3,V Vertex7/Vertex6
20:51:09  <ElementMap> <string>(1): duplicate element mapping 'g6v2;SKT;:H122e,V -> g6v2;SKT;:H122e,V;D1;:H122e:3,V Vertex8/Vertex1
```

</p>
</details> 

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 (Git)
Build type: Release
Branch: feature/free-form
Hash: e478192aa5393de9b7df190341a043f79ec6ac49
Python 3.11.8, Qt 5.15.8, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-14:
No errors on my side.
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 (Git)
Build type: Release
Branch: main
Hash: e478192aa5393de9b7df190341a043f79ec6ac49
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.50
  * fasteners 0.5.29
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

jnorre, 2024-10-14:
No errors on Ubuntu:
```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38926 (Git) Snap 1161
Build type: Release
Branch: main
Hash: a9bff78974f301100134a13c636e23aa91be5bcf
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

hyarion, 2024-10-14:
That's strange.

And to be sure it's not my configration files, I've tried it with the [safe-mode](https://github.com/FreeCAD/FreeCAD/pull/16858) as well, still same issue.
<details><summary> 1.1.0dev.38947 +6 (safe-mode)</summary>
<p>

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38947 +6 (Git)
Build type: Release
Branch: safe-mode
Hash: a2ee0ce76d9c3a9b9198a6dea98e39ae84d5ead2
Python 3.11.8, Qt 5.15.8, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

</p>
</details>

CalligaroV, 2024-10-14:
@maxwxyz, yes of course, I can reproduce it at the first try.

However I don't think this is an issue as those aren't **errors** (FreeCAD telling you "I can't / wont / don't know how to perform the operation you requested") but simple **warnings**.
https://github.com/FreeCAD/FreeCAD/blob/facd8ca6e0303cf96297a7a7398185770c1a1e0b/src/App/ElementMap.cpp#L754
And as we can all see the operation is actually performed correctly (a rectangular-based prism is generated).

The reason of the message is that we're selecting single lines with shared vertexes and the TopoNaming logic sees that and inform the user (if the warnings are **enabled**).

I tried to do the same with LS3 but no warnings are generated, probably because in that fork the Pad operation first tries to convert the lines into a Wire, and in any case the code for the warning is there too
https://github.com/realthunder/FreeCAD/blob/a9810d509a6f112b5ac03d4d4831b67e6bffd5b7/src/App/ComplexGeoData.cpp#L1747

IMO things should stay as they are: in case of future issues (operations not performed because of duplicated elements) this kind of messages will definitely help developers to find out the cause(s).

In this case this issue to me looks more something like "It's working but I don't like to see those messages" and personally I don't think I'll put efforts in solving it.
I will if some error (caused by the logic, rather that the usage) will be reported in the future, and, as I wrote above, having those messages will help me (or anyone else trying to solve those)

Side note: there was some discussion about this kind of workflow (extruding only a subset of a geometry)
https://forum.freecad.org/viewtopic.php?p=773117#p773117
In
```ruby
[code]
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * Assembly4 0.50.15
  * freecad.gears 1.2.0
  * Beltrami 1.1.0
  * sheetmetal 0.4.24
  * Curves 0.6.44
  * fasteners 0.5.26
  * kicadStepUpMod 11.2.7
  * OpenTheme 2024.8.10
  * Render 2024.8.4
  * Ondsel-Lens 2024.7.5.02
[/code]
``` 
you can't do something like this

hyarion, 2024-10-14:
Thanks for looking into it.

As a user I don't know that these red messages are safe warnings that is ok for me and helpful for toponaming developers.

I think this is an UX issue.
When I saw these I didn't know if I entered an undefined buggy state that will cause issues down the line.

Would it be ok to:
* hide these for normal builds (maybe configureable in settings or on compile time with -DSHOW_DEBUG_TOPONAMING or similar)
* or prepend "Warning"
* or at least change color for these to orange?

mwganson, 2024-10-14:
I do not get any warnings or errors here:

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default

Partial sketch padding, while supported by the software, is not a good workflow, IMO.  It makes the model more difficult to read weeks or months later when it is revisited for further editing.  Better to use a subshapebinder of those selected edges as this more clearly conveys what is being done.  (The binder should be claimed as a child of the Pad for greater clarity, but that is another issue.  Only those objects derived from PartDesign::Feature should be excluded from claimChildren(), such as when a face of a feature is padded.)

CalligaroV, 2024-10-14:
You're welcome @hyarion!

Yes, also to me this looks more a UX issue than a TopoNaming one.

It may be the beginning of a buggy state if at a certain point during modeling the two endpoints are separated after something that relies on them is built, generating therefor some kind of uncertainty about which point should be kept as reference, but also in that case I can't tell without a working example if the logic will handle it well or not.

Your suggestion about adding/wrapping those messages in a precompiler directive makes sense to me.
It may be enough to wrap them in the already existing `FC_DEBUG` (and this may also solve #14518)

Besides that, a non-developer user already can disable the warnings with the setting in the Preferences Page https://wiki.freecad.org/Preferences_Editor#Report_view but I can imagine that some warning may be useful.

About differentiating errors and warnings, IIRC if you have the Notifications enabled (https://wiki.freecad.org/Preferences_Editor#Notification_Area) the warnings have the icon :warning:, while the errors have the icon :x:

Any case, as wrote before, ATM I don't think I'll actively participate in the development of this kind of modification, there are other topics/issues I'm more "suited" for :sweat_smile:.
But if you or someone else want I'll be happy to participate to the discussion (if I have enough time) :wink:

Thanks also to @mwganson for contributing to the discussion!
As I wrote in another issue/PR, in this case I'm a terrible FreeCAD user as I don't remember the last time I used the (Sub)ShapeBinders, therefor I think I won't be much of help with that topic.

Sorry!