FreeCAD/FreeCAD #17235: Part: Ruled surface is incorrectly created on planes other than XY

FEA-eng, 2024-10-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Ruled surfaces seem to be severely broken when created on a plane other than the XY plane.

This issue mentions offset and some other operation working only on the XY plane: #15675

But it turns out that the ruled surface itself is also broken: #16979

It may not be visible right away (rectangular surface case - shows up when using it for FEM) but [this example](https://github.com/FreeCAD/FreeCAD/issues/16979#issuecomment-2392343565) is broken from the start if not done on the XY plane.

IMO it's a really nasty issue, especially since it can be hidden initially. Might be good to mention it in the documentation for now but I hope it can be fixed in the near future.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 (Git)
Build type: Release
Branch: main
Hash: e478192aa5393de9b7df190341a043f79ec6ac49
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-10-14:
I've at least narrowed down the regression to a week.

Good build:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git) AppImage 28th June 2024
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
```

Bad build:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage 5th July 2024
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
```

FEA-eng, 2024-10-14:
@Syres916 Thanks, 38334 was indeed already reported as broken.

@maxwxyz This issue is a pretty serious regression. Perhaps a blocker?

Syres916, 2024-10-14:
Root cause is https://github.com/FreeCAD/FreeCAD/commit/d9ba737132696f761aeba3bc0f1c03de8aa30df4 the code being removed as it caused a massive performance (minutes instead of seconds) impact loading models see https://github.com/FreeCAD/FreeCAD/issues/14952
So whether there can be a specific condition applied to certain _surface_ objects without impacting any _solid_ objects that triggers the Shape fix code??

Syres916, 2024-10-14:
It's taken me a while of debugging but it appears the shape that actually needs _fixing_ is the Sketch, the Ruled Surface is only doing what it's told so the invalid info is coming from Sketch.

Any fix code to correct just sketch objects causes a significant performance impact on loading so I'm going round in circles on this.
What really needs investigating is how such a simple sketch is getting in an invalid state on a non X-Y plane in the first place.

@CalligaroV could you add this to your (long) list please, I'll write a few tests that work correctly in 0.21.x in preparation for your work, most ruled surfaces are created using two open wires so this is a slightly niche case IMHO but as far as I'm aware a ruled surface can be created at any bizarre angle not just on main planes. The changes I made to carry out the testing can be seen [here](https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Testing_Ruled_Surface_Error)

CalligaroV, 2024-10-14:
@Syres916, Thanks for the ping!

Sure, I'll add it to my list!
However ATM I can't give any guarantee about how much time I'll need (end even if I'll be able) to find a good fix, surely the code you wrote will be a good starting point.

As this is a Blocker I think that this issue will by my focus for this week unless @maxwxyz and/or @chennes prefer me to give priority to something else

FEA-eng, 2024-10-15:
> It's taken me a while of debugging but it appears the shape that actually needs _fixing_ is the Sketch, the Ruled Surface is only doing what it's told so the invalid info is coming from Sketch.

Then the issue can be worse than I thought since even a sketch of a rectangle causes this. No need for a slot (which can be broken too: #17074).

Syres916, 2024-10-15:
When looking at the output from the similar area of code in 0.21.x it shows that there should be two `Wires` being passed (Sketch Edge1 and Sketch Edge3) but when I was carrying out the tests yesterday and checked for different Topo types it was returning `TopAbs_FACE` twice before the fix was applied, hope that helps.

Syres916, 2024-10-15:
@CalligaroV apologies for the delay, here are the tests that run successfully on 0.21.x with no obscure representations (hence why I've disabled the teardown so you can view the surfaces) : https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Part_Ruled_Surface_Tests

CalligaroV, 2024-10-15:
@Syres916 no need to apologize.

I'm the one who should apologize, haven't yet started looking at the code, only red the last comments and thought about possible ways to fix this issue.

I'll start by looking at the test you setup, then the `fix()` and related method's code in LS3, at least to have something more solid to use as base for some logic.

My hopes are that something may be done using OCCT's `ShapeFix` package methods.

CalligaroV, 2024-10-18:
Hyper-quick update:

Tested also with LS3 fork @
```ruby
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 2024.409.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20240407stable)
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: Italian/Italy (it_IT)
Installed mods: 
  * Assembly4 0.50.15
  * freecad.gears 1.2.0
  * Beltrami 1.1.0
  * sheetmetal 0.4.24
  * Curves 0.6.44
  * fasteners 0.5.26
  * kicadStepUpMod 11.2.7
  * OpenTheme 2024.8.10
  * Render 2024.8.4
  * Ondsel-Lens 2024.7.5.02
```

The outcome is the same of **main**

AFAICS this should mean that `Part::TopoShape::fix()` isn't the root cause of the regression (LS3 too has that method, haven't checked line-by-line if it's the same, stopped looking at the code after seeing that the visual representation of the models generated by the tests is the same of **main**)
However this is also opposite to @Syres916's https://github.com/FreeCAD/FreeCAD/issues/17235#issuecomment-2411095632, which makes me pretty confounded.

@Syres916, can you confirm that if you perform the same test (or better, create a Ruled Surface with the `*.FCStd` files produced by your tests) in a version previous to https://github.com/FreeCAD/FreeCAD/commit/d9ba737132696f761aeba3bc0f1c03de8aa30df4 the model has no issue?

It may be an issue more related to how Sketcher WB places the elements in the 3D space rather that a Part issue (or maybe this comment of mine is just nonsense)

TIA!

Syres916, 2024-10-21:
> AFAICS this should mean that `Part::TopoShape::fix()` isn't the root cause of the regression (LS3 too has that method, haven't checked line-by-line if it's the same, stopped looking at the code after seeing that the visual representation of the models generated by the tests is the same of **main**) However this is also opposite to @Syres916's [#17235 (comment)](https://github.com/FreeCAD/FreeCAD/issues/17235#issuecomment-2411095632), which makes me pretty confounded.

Baring in mind that I wrote my comments as I saw the feedback at that specific time, I'll try to clarify, https://github.com/FreeCAD/FreeCAD/commit/d9ba737132696f761aeba3bc0f1c03de8aa30df4 (June 29th 2024) is the commit that brings the symptom to the user but I completely agree that commit isn't the root cause.

I've just compiled it's parent commit https://github.com/FreeCAD/FreeCAD/commit/3b8f615f7a90805682fed6a306510a041a69cca8 and while the surfaces are flat they are not displaying correctly as if one edge has been rotated 180 degrees about it's centreline.

On May 18th 2024 as part of https://github.com/FreeCAD/FreeCAD/commit/2657bbee4d72df816438f9d9afd7ba860ace6001 the code accidentally covering up the actual issue was added but at that time the compiled weekly builds were still using non-TNP code, the TNP CMake change https://github.com/FreeCAD/FreeCAD/commit/895803a8cb9e4a8a8757088ad219f6eb9db3f197 was trigged on May 20th 2024.
So the only way IMHO the actual root issue can be determined is to _git bisect_ with _-DFC_ENABLE_TNP_FIX=On_ from (educated guess) early February time up to commit 2657bbee4d72df816438f9d9afd7ba860ace6001

As this git bisect will consume my machine all day, I won't be able to do this until tomorrow.

Syres916, 2024-10-22:
Even though many hours has been consumed, very little progress made with the bisect, going to be a long night.

Syres916, 2024-10-23:
I've had to call it a day, wasted all night trying to fix uncompilable commits so someone else will have to try to find the commit. Just a note if no-one noticed but you can't use the CMake command line flag to enable TNP,  I had to change the end lines of _SetGlobalCompilerAndLinkerSettings.cmake_

luzpaz, 2024-10-23:
@Syres916 thank you so much for your efforts. Please make sure not to overwork or burn yourself out.

CalligaroV, 2024-10-23:
@Syres916, I thank you too as @luzpaz already did.
Twice for the hint you gave in your previous comment.

I think this is closely related to https://github.com/FreeCAD/FreeCAD/commit/2657bbee4d72df816438f9d9afd7ba860ace6001: all the Ruled Surfaces generated with your extremely useful test cases have edges laying on the XY plane (which IIRC should be the default plane for any newly created element).

I did some tests with modified code in the last couple of days and I can kind of confirm this:
adding, only for debugging purposes only, new variables like `double x {testShape.Location().Transformation().TranslationPart().X()};` for all the translation and rotation parts of a shape (`TopoDS_Shape`, `TopoShape`), after `TopoShape::makeElementRuledSurface()` those parts are all always `=0` exept for `double w {ruledShape.Location().Transformation().GetRotation().W()};` which value is `=1`

I'm still trying to figure out at what point of the code this is happening and why.

The strange (but probably useful) thing is that if I try to create a Ruled Surface with 2 separate sketches (containing only one part each) then the Ruled Surface generated looks correct (a simple slot-like face).

I'll keep searching tomorrow and in the next days.
If you don't see other comments from me then probably it's simply because I'm lost in code/tests :innocent:

maxwxyz, 2024-10-24:
Is https://github.com/FreeCAD/FreeCAD/pull/17436 fixing this issue?

FEA-eng, 2024-10-24:
> Is #17436 fixing this issue?

Yes

closed 2024-10-25 by chennes

chennes, 2024-11-06:
Noting that @mwganson is eligible to claim the bounty from the FPA for PR #17436. To claim, please send an email to fpa@freecad.org referencing this issue, and your preferred method of payment.