FreeCAD/FreeCAD #17239: RC2 step export misfire

JaapStruyk, 2024-10-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

drawing not exported as expected, draw a square tube, cut the end trough all and on the export the cut isn't trough all but 5mm, adding a cut of whatever kind after that and it's as it should be.
[image]

### Full version info

```shell
OS: macOS 11.7.10
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * Manipulator 1.5.7
  * sheetmetal 0.4.13
  * Curves 0.6.36
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-10-14:
I think this is caused by a bug in the PartDesign Workbench: #17210. Please recompute the whole document and try exporting again.

JaapStruyk, 2024-10-15:
Recompute indeed solves the export bug...

luzpaz, 2024-10-22:
This is a UI issue. Should we address it ?

Roy-043, 2024-10-23:
This has nothing to do with the GUI AFAICT. But the issue mentioned in my previous comment has been fixed.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

closed 2024-10-23 by Roy-043