FreeCAD/FreeCAD #17241: random crash of the software

CarloSartori, 2024-10-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

During the normal use of the program, mostly with Part Design workbench, the program froze and than crash. I can not replicate the issue since sometimes when i redo the same steps it works fine. Sometimes it crash every few minutes, sometimes it is stable for hours.
I'm not sure how I can better describe the issue.
these are the freecad version:

> OS: Windows 11 build 22631
> Word size of FreeCAD: 64-bit
> Version: 1.0.0RC2.38806 (Git)
> Build type: Release
> Branch: (HEAD detached at 1.0rc2)
> Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
> Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
> Locale: Italian/Italy (it_IT)
> Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
> Installed mods: 
>   * CADExchanger
>   * fasteners 0.5.29
>   * freecad.gears 1.3.0
>   * sheetmetal 0.4.26

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * CADExchanger
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * sheetmetal 0.4.26
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-14:
Thanks for reporting but we do need steps to create the crash.  
Perhaps, you could  try using the 1.1-dev weekly version to see if you can replicate ?   
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

CarloSartori, 2024-10-14:
I'm downloading right now. I'll attach a log at the next crash (hopefull not too soon :P) how can I generate the log? do I have to activte something?

luzpaz, 2024-10-14:
You can invoke FreeCAD from a command line terminal or console  
`FreeCAD.exe --log-file <path\to\file.log>`

CarloSartori, 2024-10-15:
I'm sorry but I'm not able to start freecad with --log-file option. if I try on windows CMD I had this error:
`Sintassi del comando errata.`
tha is italian for syntax error
if I try freecadcmd I have this answer:
```
  File "<stdin>", line 1
    --log-file <C:\Users\Admin\Documents\log\FC1.0.log>
                   ^
SyntaxError: unexpected character after line continuation character
```

edragos, 2024-10-15:
Hello!
I also encountered this.
I believe that is almost impossible to replicate by the developers.
Why?
About two weeks ago, I worked in FreeCad almost 4 hours and everything was fine.
The entire work was done in Part Design.
I have made a pause.
Then, I open FreeCad, New Document and I draw a fully constrain rectangle.
Close Sketch, Pad, Ok.
While I’m looking at that form, the mouse cursor changed to a rotate circle and FreeCad closed itself.
Under 2 minutes.
I recommend, if this happen again, go to Event Viewer, Windows Logs, Application and look what it says.
My feeling is that is more a Windows issue, than FreeCad. But is only my guess.
I turned off Memory integrity and for some time was ok.
In the last weekend I proceed a fresh installation of Windows(fresh, fresh boot from usb, format hdd) and till now everything is fine with Memory integrity turned on.
So, I don’t know why that happened.

Many thanks to this wonderful community.

luzpaz, 2024-10-15:
@edragos we need more data from you. 
* On what version did you have issues? (Full [About](https://wiki.freecad.org/About) info)

luzpaz, 2024-10-16:
@CarloSartori any luck?

CarloSartori, 2024-10-17:
Hi to all. I used the dev version all day long yesterday without any problem. I'll try again today. Unfortunately I'm not able to replicate the problem. it just happens.

regarding the info I have no idea where to find more information than the one I already posted.
The version I'm using now is:

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 (Git)
Build type: Release
Branch: main
Hash: e478192aa5393de9b7df190341a043f79ec6ac49
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * CADExchanger
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * offline-documentation 1.0.0-alpha
  * sheetmetal 0.4.26
```

luzpaz, 2024-10-17:
@CarloSartori no worries. Thank you. For now I will close this ticket as to avoid bugtracker bloat (we don't have enough info here to reproduce). But of course, if you can reproduce the bug feel free to return and notate this ticket and we can re-open it. Thank you for your contribution nonetheless.

CarloSartori, 2024-10-24:
I hoped to never came back here, but here is the log of the crash I had this morning

Thanks
Carlo
[fc2410.log](https://github.com/user-attachments/files/17502659/fc2410.log)

luzpaz, 2024-10-24:
> I hoped to never came back here, but here is the log of the crash I had this morning

@CarloSartori thanks for reporting back. Please remember that you help us out a great deal when you paste your full [About](https://wiki.freecad.org/About) info on the version that you reproduce the crash on

hyarion, 2024-10-26:
I don't see any info in the log about a crash

luzpaz, 2024-10-27:
Indeed. @CarloSartori the log provides no insight into the issue.

hyarion, 2024-10-28:
It seems like the log files doesn't have crash information in them.
Does anyone know how to get this on windows?

luzpaz, 2024-10-28:
>  Version: 1.1.0dev.**38946** (Git)

@CarloSartori you still aren't using the most up to date weekly build for windows

[image]

**39057** is the latest revision in https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds ATM

edragos, 2024-10-29:
I’ve suggested to Carlo, two weeks ago, to post together with the FreeCad log and Windows Application Log related to FreeCad.
I’ve recommended this because when an application suddenly crash, it can no longer write in its log.

luzpaz, 2024-10-29:
> @edragos we need more data from you.
> 
>     * On what version did you have issues? (Full [About](https://wiki.freecad.org/About) info)

2 weeks ago I asked you for this (https://github.com/FreeCAD/FreeCAD/issues/17241#issuecomment-2415038393)  
We're trying to help both of you but the process of getting relevant info from you both is very slow.  There are methods of deducing and troubleshooting but we need engaged cooperation. 
* The highest priority needs is your full [About](https://wiki.freecad.org/About) info.
* Then we can ask you to run the program from your command line console (so you can see errors when a crash happens). 
* You can start FreeCAD in Safe Mode (starts FreeCAD with temporarily new config files without any addons enabled).   
* Output a log file when running FreeCAD

edragos, 2024-10-29:
I don’t have any issues.
This is the reason that I didn’t put the About info.
I know that About info is the first thing when a problem is reported.
But I have no problems. I’ve made only a suggestion in hope to help Carlo.

OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Europe (en_150)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

CarloSartori, 2024-11-07:
hi to all
I'm sorry if I'm not engaged enough. 
I would love to help you since you are doing such a great work, but htat is what I can do, take it or leave it.
 I will install the last update available and hopefully this will stop the issues (taht are still happening)
If the issue will happen again (i hope not) I'll post the full about info, the frecad log and the windows log.
In the meantime thank you for your help and your amazing job.

luzpaz, 2024-11-08:
Thanks @CarloSartori and @edragos 

Unable to reproduce. So we'll close this. If any new insights emerge on how to trigger this bug or more clues, please notate this ticket and we'll re-open it.

closed 2024-11-08 by luzpaz

CarloSartori, 2024-11-11:
Hi to all
I have some information to add, hoping them could be useful
I had a crash and I checked the windows event viewer.
Here the message (sorry it is in Italian)

`Nome dell'applicazione che ha generato l'errore: freecad.exe, versione: 1.1.0.0, timestamp: 0x67246489
Nome del modulo che ha generato l'errore: CoreMessaging.dll, versione: 10.0.22621.4391, timestamp: 0x373137e2
Codice eccezione: 0xe0464645
Offset errore 0x000000000008405d
ID processo che ha generato l'errore: 0x0x5210
Ora di avvio dell'applicazione che ha generato l'errore: 0x0x1DB343A4BE89EC0
Percorso dell'applicazione che ha generato l'errore: C:\Program Files\FreeCAD 1.0 RC\bin\freecad.exe
Percorso del modulo che ha generato l'errore: C:\WINDOWS\SYSTEM32\CoreMessaging.dll
ID segnalazione: 1cc921bf-4e0d-48ab-b2f7-c04b582c88bd
Nome completo pacchetto che ha generato l'errore: 
ID applicazione relativo al pacchetto che ha generato l'errore: `

in short terms, the problem is due to coremessaging.dll

here the freecad full information
`OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * CADExchanger
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * offline-documentation 1.0.0-alpha
  * Render 2024.9.25
  * sheetmetal 0.4.26
`

hope this help

CarloSartori, 2024-12-02:
Hi. I'm sorry to reopen a dead thread but the problem is still there.
I found a solution: if I set-up the program to run in compatibility mode as windows 8 it never freezes, but it is very very slow.
Unfortunately I can not use it either way, with random crash or very slow.

Thank you again for your help.

Carlo