FreeCAD/FreeCAD #17242: macOS (arm64) 1.0RC2 crashes when selecting

konart, 2024-10-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD crash on my macOS 15.0.1 (24A348) whenever I click (left or right mouse button - does not matter) any axis in a model window in a newly created file (Parametric Part).

### Full version info

```shell
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-10-14:
Which [Mouse navigation](https://wiki.freecad.org/Mouse_navigation) do you use?

hyarion, 2024-10-14:
This was one of unhandled-exception-reaches-qt-crashes which was fixed here: https://github.com/FreeCAD/FreeCAD/pull/16906

The reason for the unhandled exception has also been fixed. Latest weekly doesn't have this issue

@konart In case you want to try weekly on macOS you need to remove the quarantine check:
1. Install weekly to /Applications
2. in Terminal.app, run `sudo xattr -r -d com.apple.quarantine /Applications/FreeCAD.app`

closed 2024-10-15 by chennes