FreeCAD/FreeCAD #17243: Wire is not closed (Sketch issue)

eagabriel, 2024-10-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Almost everytime I'm facging wire is not closed issues with sketches. Even very simple ones, mostly when using trim tool.
[image]

The point indicating that is constrained coincident with the curve

[image]

If I try to reconstrain it gives me a redundant constraint

[image]

If a delete the constraint an try to redo the constraint it not connect but marking as contraint.

[image]

If I try to connect with a line it not connect and make a small line.

[image]

[image]
[IrrigationAngleAdjust.zip](https://github.com/user-attachments/files/17364416/IrrigationAngleAdjust.zip)

The sketch is attached.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Dark/Qt default
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * A2plus 0.4.65
  * Assembly4 0.50.15
  * fasteners 0.5.24
  * MeshRemodel 1.10.25
  * Render 2024.8.4
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-10-14:
I assume that this is because the circle is constructed by straight line segments. (But this is not certain)
[image]

Nuck-TH, 2024-10-14:
To pad, you need non-intersecting closed perimeters. In presented case, you need to remove part of circle between that two points(replace outer circle with arc).

Nuck-TH, 2024-10-14:
> I assume that this is because the circle is constructed by straight line segments. (But this is not certain)
no, what is on screen is not relevant - internally(in cad core) circle is mathematical circle.

kaiwas, 2024-10-14:
> To pad, you need non-intersecting closed perimeters. In presented case, you need to remove part of circle between that two points(replace outer circle with arc).

And in this case it would be great to have non-destructive cropping. I made a proposal in this thread
https://github.com/FreeCAD/FreeCAD/issues/16243

But there is no such possibility yet.

kaiwas, 2024-10-14:
> no, what is on screen is not relevant - internally(in cad core) circle is mathematical circle.

What I wanted to say is that what we see on the screen does not correspond to what is happening mathematically, due to the very large magnification.
It seems that there is no constraint, but in fact there is.

It just seems that this is not the case because the circle is shown as straight line segments.

https://github.com/user-attachments/assets/9ca173c3-38e9-4c50-802c-4a607966f7bd

In general, I think that this is a problem with displaying rounded objects due to their on-screen segmentation.

eagabriel, 2024-10-14:
I think this wire is not closed thing should have a tolerance, because the fact is even in mathematical internally it is connected when I check it it say that it is not.

Nuck-TH, 2024-10-14:
> > To pad, you need non-intersecting closed perimeters. In presented case, you need to remove part of circle between that two points(replace outer circle with arc).
> 
> And in this case it would be great to have non-destructive trim. I made a proposal in this thread #16243
> 
> But there is no such possibility yet.

When line attached to circle "Trim edge" does exactly what needed - it turns circle into arc.

mwganson, 2024-10-14:
User error, not a bug.  The wire is not closed error is because there are t-sections in the sketch.

The appearance that the edges are not connected is an illusion.  The object must be tessellated in order to be displayed.  Select the sketch and switch to the view tab in the properties and adjust the Deviation property from 0.5 to 0.1 to see the edges now appear closer than before.  This is because now a finer tessellation is produced for display purposes (at the expense of more CPU cycles and poorer performance in recomputing).

You must remove the line segment between the 2 points shown in red in this image to resolve the wire not closed error message.
[image]

eagabriel, 2024-10-14:
ok it works, tks

closed 2024-10-14 by luzpaz

dzid26, 2025-01-01:
> To pad, you need non-intersecting closed perimeters. In presented case, you need to remove part of circle between that two points(replace outer circle with arc).

> User error, not a bug. The wire is not closed error is because there are t-sections in the sketch.
> 
> You must remove the line segment between the 2 points shown in red in this image to resolve the wire not closed error message. ![Snip macro screenshot-836a41](https://private-user-images.githubusercontent.com/39143564/376376762-8ed3c1a6-0d0d-4b91-b057-cf21c02934fb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU3NjIyOTYsIm5iZiI6MTczNTc2MTk5NiwicGF0aCI6Ii8zOTE0MzU2NC8zNzYzNzY3NjItOGVkM2MxYTYtMGQwZC00YjkxLWIwNTctY2YyMWMwMjkzNGZiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTAxVDIwMDYzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFhODI3MWEyODk4NjBjNGE5Y2QzOWFiY2Q1MmIwNjg3OGVjNzYzYzYwODZiY2NjMjQ0YWE3MWFhYTEzY2NkNzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.9t-3O0iSViKc-mw08ANgZZ-HEzWr_4QamA_VhSMj5fU)

Ideally, this segment should be shared between two perimeters and two pads would be merged - it is similar when using `Allow Compound`, except without a gap.

If there was an actual intersection it would not be clear which geometry is outer/inner, but here when there are only T-sections we can always consider those perimeters additive.  

So there is a room for improvement. Especially useful, when people start to use "master sketches" with multiple geometries.