FreeCAD/FreeCAD #17247: Assembly: Dragging does not auto-recompute (e.g. measurements)

maxwxyz, 2024-10-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

from https://github.com/FreeCAD/FreeCAD/issues/14559

Dragging/moving the assembly does not trigger an automatic recompute which can lead to outdated objects which need a manual recompute (e.g. measurements).
Automatic recompute should be handles with care so there is no issue with large files and assemblies if auto-recompute would be active.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37668 (Git) AppImage
Build type: Release
Branch: main
Hash: ea68c5e88cdc80044ede482f6ad908820f4c1c82
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * freecad.gears 1.2.0
  * A2plus 0.4.64a
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * Trails 2022.1.0
  * PieMenu 1.2.7
  * fasteners 0.5.21
  * Render 2024.1.23
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MiqCG, 2025-04-13:
Dear, @maxwxyz ,
I was thinking in opening a request feature to add the option of automatic recompute of assemblies, as it is tedious to recompute after every inserted component or joint creation/modification. As you said, not recomputing the assembly "can lead to outdated objects" that make harder to find the conflicting step afterwards.

Searching for similar issues, your text has make me doubt if already exists the option to check the automatic recompute of assemblies. Haven't found it in the Preferences > Assembly > General.
To avoid performance problems with large assemblies, my idea is to propose that FreeCAD has the Preference option to check to "Auto-recompute if less than N joints and/or M inserted components", where the user could decide what are N and/or M depending on their needs and computer capabilities.

maxwxyz, 2025-04-13:
@PaddleStroke is this possible and would it make sense to limit on the number of joints?