FreeCAD/FreeCAD #17250: Start: recent file with accent in filename is not open

FlachyJoe, 2024-10-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Clicking the recent file results in an error dialog
**The file '/home/florian/T\xe9l\xe9chargements/Body update/simple-test.FCStd' cannot be opened**

```bash
:~$ ls -l /home/florian/Téléchargements/Body\ update/simple-test.FCStd 
-rw-rw-r-- 1 florian florian 14251 14 oct.  21:00 '/home/florian/Téléchargements/Body update/simple-test.FCStd'

```

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (XFCE/xfce/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Debug
Branch: dev
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.12.6, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.2.dev
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

It worked on 1.1.0dev.38923

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-14:
Another unicode path bug

Syres916, 2024-10-14:
@FlachyJoe I'm assuming this regression was introduced today, https://github.com/FreeCAD/FreeCAD/commit/777e2c7a80bc8fff155e565eb99fbfd2912bc07d ?

hasecilu, 2024-10-17:
I was about to report the same, in my case it was the name that use Unicode fractions glyphs. File menu is OK.

[image]

PR #17265 (2024-10-21): Core: Add convenience class ModuleIO to open files
* Use ModuleIO in RecentFilesAction & RecentMacrosAction
* Use ModuleIO in MainWindow::loadUrls
* Use ModuleIO in StartView::fileCardSelected

Fixes #17250
closed 2024-10-21 by yorikvanhavre