FreeCAD/FreeCAD #17254: Model jumps around wildly on screen when rotating

theosib, 2024-10-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm on a Mac, and I have touchpad mode enabled for user input.

When I hold down Option to rotate this model and move my finger on the trackpad, it jumps around wildly as long as the ShapeString is visible.
[24-sided-letter-die.FCStd.zip](https://github.com/user-attachments/files/17369888/24-sided-letter-die.FCStd.zip)

### Full version info

```shell
OS: macOS 14.7
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 (Git)
Build type: Release
Branch: main
Hash: e478192aa5393de9b7df190341a043f79ec6ac49
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * Pyramids-and-Polyhedrons
  * fasteners 0.5.29
  * Curves 0.6.50
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-14:
@johnocfii would you mind testing this out on macOS? TIA

hyarion, 2024-10-14:
Lol, I can reproduce this:

1. load file
2. hover over part so the rotation anchor is on the object
3. rotate

https://github.com/user-attachments/assets/95b3f63e-0d52-4d48-bd95-aa014553c56c

The model looks a bit broken when shape string is visible:
[image]

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 +1 (Git)
Build type: Release
Branch: wip
Hash: 077596644ad00414a37c42633965de3d3b758e50
Python 3.11.8, Qt 5.15.8, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

hyarion, 2024-10-15:
This happens in multiple navigation styles and zoom is also affected a bit

chennes, 2024-10-16:
I don't understand how this meets the criteria to be a blocker -- there is no crash, and no loss of data. I'm not arguing it shouldn't be solved, but at this late date we need to be very VERY clear about how blockers are chosen, and I don't believe this is one.

hyarion, 2024-10-16:
I just looked at this again and saw that the reason why this file fails is that the shape string is placed 703,42km to the side (403 miles in freedom units). The shake starts to be noticeable at around 5km offset.

The issue can also be seen if two standard cubes are created and one moved 100km to the side and you try to rotate the camera around the object at the origin.

Looking at op's 24sided die in the showcase channel over at discord, I assume this was a previous attempt where op accedentally clicked k instead of m when entering a number.

While there obviously is an issue with a shake of about 1/1,000,000 of the model you're creating, I don't think this is a blocker.

As long as we don't get a real use case where this is a problem (modeling ventilation on a [death star](https://github.com/FreeCAD/FreeCAD/pull/17214)?) I don't think the camera shaking is an issue. However, maybe the placement input fields should be limited to 1,000 times the size the objects bounding box to help the user from making this type of mistake?

Or we could just close this as not-planned.

maxwxyz, 2024-10-16:
is this the same or related to https://github.com/FreeCAD/FreeCAD/issues/12162 @Rexbas

Rexbas, 2024-10-16:
When the object is placed 700km away then the object center is far away causing the object to go out of view when you start rotating with the mouse not over the object. If the mouse is over the object and you start rotating then the shaking is likely caused by floating point precision issues because the near and far clipping plane values are very big.

So yes this is the same as #12162.

maxwxyz, 2024-10-16:
Duplicate of #12162

closed 2024-10-16 by maxwxyz