FreeCAD/FreeCAD #17255: FEM: crash or freeze RC2 when meshing before save

fandaL, 2024-10-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
1. In the Preferences -> FEM -> General -> set "Beside .FCStd file"
2. Open New file, create a solid, create a mesh with GMSH and click "Apply"

It appears:
`Can't start Solver or Mesh creation besides FC file` - name of the error prompt
`Please save the file before executing a solver or creating a mesh. This must be done because the location of the working directory is set to "Beside *.FCStd File". For the moment the tmp dir /tmp/fcfem_p9u_94wx is used.`
`QObject::setParent: Cannot set parent, new parent is in a different thread`
That is probably correct error, however, FreeCAD crashes immediately (I recorded a video to be able to see an error message) on Ubuntu. On Windows, FreeCAD freezes (so I was able to read the error message).

Of course, issue is solved when a file is saved before meshing, but I think FreeCAD should not crash or freeze.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38641 +165 (Git) Snap 1154
Build type: Release
Branch: tag: 1.0rc2
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Defeaturing 1.2.2
-------------------------------------------------------
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: Czech/Czech Republic (cs_CZ) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Defeaturing 1.2.2
  * nurbs
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-14:
cc @FEA-eng

FEA-eng, 2024-10-15:
I can confirm it and I agree - while the file should be saved first, FreeCAD shouldn't react by freezing or crashing otherwise.

@marioalexis84 FYI

wwmayer, 2024-10-15:
The error dialog also appears in older versions (tested with v0.21) but the crash is a regression in v1.x

maxwxyz, 2024-10-15:
Labelled as blocker for the regression and easy to reproduce crash.

PR #17278 (2024-10-16): Fem: Do not use main window widget from another thread - fixes #17255
Fixes #17255.
Now the mesh processes are running in another thread. Therefore, calling widgets directly from main thread results in undefined behavior.

@wwmayer
commit a22f1241 (2024-10-16): Fem: Do not use main window widget from another thread - fixes #17255
commit 42d0dfa1 (2024-10-16): Fem: Do not use main window widget from another thread - fixes #17255
closed 2024-10-16 by chennes