FreeCAD/FreeCAD #17261: Fillet with tapered ends  breaks the geometry

MBratbo, 2024-10-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If a fillet follows a contour where the isn't space for the entire fillet at the ends it breaks the geometry
[image]

### Full version info

```shell
Freecad 1.0 RC2
```

### Subproject(s) affected?

None

### Anything else?

[Ny ZIP-komprimeret mappe (2).zip](https://github.com/user-attachments/files/17373945/Ny.ZIP-komprimeret.mappe.2.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-15:
Please add the full [About](https://wiki.freecad.org/About) info to the issue (`Help > About FreeCAD > Copy to clipboard`).

MBratbo, 2024-10-17:
Issue: Freecad 1.0 RC2

MBratbo, 2024-10-17:
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Danish/Denmark (da_DK)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4 0.50.2
  * CfdOF 1.27.10
  * Help 1.0.3

closed 2026-02-21 by maxwxyz