FreeCAD/FreeCAD #17266: Regression: Error <Sketch> SketchObject.cpp(11404): invalid subname

Syres916, 2024-10-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When sketching on a face if a vertex to vertex (origin) dimension constraint is created when each vertex is selected the Report View generates errors:

```
<Sketch> SketchObject.cpp(11404): invalid subname ;g1v1;SKT.Vertex1
<Sketch> SketchObject.cpp(11404): invalid subname ;g1v1;SKT.Vertex1
<Sketch> SketchObject.cpp(11404): invalid subname ;g1v1;SKT.
<Sketch> SketchObject.cpp(11404): invalid subname ;g1v1;SKT.
<Sketch> SketchObject.cpp(11404): invalid subname ;g1v1;SKT.Vertex1
<Sketch> SketchObject.cpp(11404): invalid subname ;g1v1;SKT.Vertex1
<Sketch> SketchObject.cpp(11404): invalid subname ;g1v1;SKT.
<Sketch> SketchObject.cpp(11404): invalid subname ;g1v1;SKT.
```

This was tested with existing config files and with clean config files but makes no difference. See forum topic https://forum.freecad.org/viewtopic.php?p=787131&sid=5713b65f2965c4322ac1983e35ba9077#p787131

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-10-15:
Triggered by commit: https://github.com/FreeCAD/FreeCAD/commit/38c6d842f2d2fc252bff2c9e0f43a706f24cdc48

Syres916, 2024-10-15:
I'm rapidly realising this is beyond me, in order to not completely waste the time spent on this, the vertex index of the circle centre is always `-1` hence the error thrown. It 'feels' like there should be a `rebuildVertexIndex()` somewhere to correct this but where specifically this should go even if I'm correct, I don't know.

CalligaroV, 2024-10-15:
As I usually do in similar cases (issues that may be related to TopoNaming code), I performed the following checks:

1. Try to reproduce in LS3 fork: in this case nothing interesting came up, no errors/warnings/messages are generated and the operation is performed correctly
2. Check differences in the code between main and LS3 and try to reproduce the issue with the imported code: in this case there actually are differences, in the methods `SketchObject::getVertexIndexGeoPos()` and `SketchObject::rebuildVertexIndex()` as well as the definition of a new member `std::map<std::pair<int,PointPos>,size_t> GeoPos2VertexId` in `Sketcher::SketchObject` (`src/Mod/Sketcher/App/SketchObject.h`) but also with those modifications the issue is still present.

However, during my local tests I noticed a thing: in LS3 there isn't the "AutoConstraints" logic, I had to manually select both the center of the circle and the root point.

Doing the same in main doesn't trigger any error/warning/message, confirmed also by a user of the forum (https://forum.freecad.org/viewtopic.php?p=787192&sid=8e2688bc6f78c231a777404b983f05de#p787192)

This makes me think that the issue is rather caused by something happening in the "AutoConstraints" logic (not giving the element as input to other methods?).

@PaddleStroke, IIRC the "AutoConstraints" logic is something you worked on.
If so can you give your opinion about what's happening in this case?
Or ping who worked on it.

Then we may be able to narrow the cause of this issue and work on a fix.

TIA!

Syres916, 2024-10-15:
@CalligaroV it's a regression because I'm not using the AutoConstraints, I'm old school using all separate constraints and it did fine until mid September!!

I think it's more to do with the difference between _pre-selection_ of the vertexes and then triggering the constraint command and the way I (and others) work (using _post-selection_) which is to trigger the constraint command first and then select the vertexes afterwards which is a perfectly valid way to work IMHO.

CalligaroV, 2024-10-15:
Sorry @Syres916, I haven't tried also that way (haven't even thought about it :confounded:)

I tried it again right now and I confirm it happens, even with the imported code from LS3.
I can confirm also that no issues raises if I perform the same operation in LS3.

I have the bad feeling that this will be the beginning of a trip in the White Rabbit's Hole :confounded:

CalligaroV, 2024-10-28:
@Syres916, finally I have some updates.

First (less important):
> I can confirm also that no issues raises if I perform the same operation in LS3.

This is wrong. Tried again and actually this issue happens also on LS3.

Tested
```ruby
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: a9810d509a6f112b5ac03d4d4831b67e6bffd5b7
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Installed mods: 
  * sheetmetal 0.5.3
  * Beltrami 1.1.0
  * kicadStepUpMod 11.3.9
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Assembly4 0.50.15
  * Render 2024.9.25
  * Curves 0.6.51
  * Ondsel-Lens 2024.10.21.01 (Disabled)
  * freecad.gears 1.3.0
``` 
[image]

Second (more important), I'm pretty sure I found the root cause and a possible fix.

---
**Cause**: as you correctly guessed, the issue is in `SketchObject::rebuildVertexIndex()`. In detail https://github.com/FreeCAD/FreeCAD/blob/a65d05c6e495038168eefcba6a1e43218fbb3a20/src/Mod/Sketcher/App/SketchObject.cpp#L9270 and similarly for ellipses in https://github.com/FreeCAD/FreeCAD/blob/a65d05c6e495038168eefcba6a1e43218fbb3a20/src/Mod/Sketcher/App/SketchObject.cpp#L9274 the code in `SketchObject::getVertexIndexGeoPos()` expects something like `g1v`**3** (`PointPos::mid` is equal to 3 https://github.com/FreeCAD/FreeCAD/blob/a65d05c6e495038168eefcba6a1e43218fbb3a20/src/Mod/Sketcher/App/GeoEnum.h#L87) but the Toponaming methods generates (IMO correct) mapped names like `g1v`**1**.

This difference causes `SketchObject::getVertexIndexGeoPos()` to return -1, triggering the error message

---
**Possible fix:** modify `SketchObject::getVertexIndexGeoPos()` as follows:
```c++
int SketchObject::getVertexIndexGeoPos(int GeoId, PointPos PosId) const
{
    for (std::size_t i = 0; i < VertexId2GeoId.size(); i++) {
        if (VertexId2GeoId[i] == GeoId
            && ((getGeometry(GeoId)->is<Part::GeomCircle>()
                 || getGeometry(GeoId)->is<Part::GeomEllipse>())
                        && VertexId2PosId[i] != PosId
                    ? PointPos::start
                    : VertexId2PosId[i])
                == PosId) {
            return i;
        }
    }
    return -1;
}
``` 
that means: if the geometry is a circle or an ellipse and the argument `PosId` isn't as expected then match it against `PointPos::start` (== 1).

I tried other ways to fix this issue but ATM this is the simplest way I found.

I don't know which side effects may have this kind of modification, therefor before opening a PR I'd like to ask @realthunder's opinion (as he too may want to integrate a fix in LS3).
Also @PaddleStroke's and/or @AjinkyaDahale's opinion should be kept into consideration as their knowledge of Sketcher is undoubtedly wider than mine.

TIA!

PaddleStroke, 2024-10-29:
Just small comments :
Why do you say it's correct to have g1v1? Wouldn't it make more sense to expect g1v3 1nd fix the other way around? I don't know tnp code so it's just a feeling, maybe you reasons are perfectly good.

- your case handle only circle and ellipse. What about arcs?

CalligaroV, 2024-10-29:
Thanks @PaddleStroke for the inputs.
About your questions:

> Why do you say it's correct to have g1v1? Wouldn't it make more sense to expect g1v3 1nd fix the other way around? I don't know tnp code so it's just a feeling, maybe you reasons are perfectly good.

I haven't an answer that links to the actual part of the code that generates the names but, looking at the OP, it looks that the actual Toponaming logic generates names with an ending **1** instead of an ending **3**.

My guess is that this number is a progressive index of the `TopoDS_Vertex`es found by the logic while "scanning" the geometries inside a sketch and, likely, for a circle and/or an ellipse it finds only 1 point (the center point).

I tried to apply modifications that set/expect and ending 3 but all of them led to other error messages, set aside that the parts of the code were several, one for each new error message.

Another reason why I haven't investigated much about how to fix it that way is simply because I haven't found an answer to the questions "why aren't there also `g1v1` and `g1v2`?" and "why if this geometry has only 1 point (ok, mathematically all of them have infinite points) it's number is **3** instead of **1**?"

Therefor I preferred to share the possible fix of my previous comment.

>     * your case handle only circle and ellipse. What about arcs?

This is simply because, in case of arcs, the center point is `g1v3` or, in any case, there are always 3 points (start, end, center/mid) and therefor a `g1v3` is (surely) foreseen.

In the code, you can see kind of proof of that in the lines that follows this one: https://github.com/FreeCAD/FreeCAD/blob/4f323f9580334de25192d2bf1f272206ad2d3f55/src/Mod/Sketcher/App/SketchObject.cpp#L9276

**Summarizing**: we can fix this the way we think is best.
Personally, I prefer the one I'm suggesting, it makes more sense to me (a circle has only 1 point)
If we choose the other way then I'll simply need more time to figure out how to develop it.

And in both cases, a minimum amount of tests (possibly manual and/or automatic) should be foreseen IMHO.

ErtugrulEge, 2025-06-22:
Greetings everyone!

`SketchObject.cpp(11321): cannot find subname ;g32;SKT.Edge3`

When sketching I get this message but I had difficulties to reproduce. In this file, when delete a dimension and make another, creates error message for me in sketcher.

I can say I get this error without sketching on a face just two lines are enough.

[vertical_dimension(1).zip](https://github.com/user-attachments/files/20852149/vertical_dimension.1.zip)

** Delete existing dimension and try to dimension the same line or other lines

[image]

[image]

d4v33y, 2025-06-22:
https://github.com/FreeCAD/FreeCAD/issues/17266#issuecomment-2442512385
---
Hi @CalligaroV, I have the same issue with "...SketchObject.cpp (...): cannot find subname ..." (or invalid subname) but I cannot modify the cpp file because I downloaded it from the software manager (flatpak). I searched the cpp file in all the pc but i couldn't find it.
You downloaded it from browser?

Furthermore, I confirm [this comment in the Freecad forum](https://forum.freecad.org/viewtopic.php?p=787192&sid=52726c87746c1ed051b55effa8daafa2#p787192). But this seems a workaround, not the very solution...

Should I reinstall it in another format?

maxwxyz, 2025-08-17:
@drwho495 should we add this to the Toponaming board?

drwho495, 2025-08-17:
> @drwho495 should we add this to the Toponaming board?

Sure

PaddleStroke, 2025-12-16:
Is this bug not a duplicate of https://github.com/FreeCAD/FreeCAD/issues/25787 which was fixed by https://github.com/FreeCAD/FreeCAD/pull/25920 ?

PaddleStroke, 2025-12-16:
Yes it is. In the forum topic the reproduction steps match 25787 exactly: dimension between origin and circle center.
@chennes @maxwxyz please close as duplicate

closed 2025-12-16 by maxwxyz