FreeCAD/FreeCAD #17268: Problem installing workbench ImportNURBS

KeithSloan, 2024-10-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Problem installing workbench ImportNURBS

When it tries to install the python dependence rhiono3dm it goes to use pip and has a problem

### Full version info

```shell
Problem installing workbench ImportNURBS

When it tries to install the python dependence rhiono3dm it goes to use pip and has a problem

Report View
19:45:08  Saved slider position at 4392
19:45:21  Attempting to fetch remote copy of ALLOWED_PYTHON_PACKAGES.txt...
19:45:21  Overriding local ALLOWED_PYTHON_PACKAGES.txt with newer remote version
19:45:55  QObject::moveToThread: Current thread (0x600000fcc0f0) is not the object's thread (0x0).
Cannot move to target thread (0x600000c19b40)

19:45:55  You might be loading two sets of Qt binaries into the same process. Check that all plugins are compiled against the right Qt binaries. Export DYLD_PRINT_LIBRARIES=1 and check that only one set of binaries are being loaded.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-15:
Can you add your full About info, please? TIA!

KeithSloan, 2024-10-16:
OS: macOS 14.4
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * ImportNURBS 1.1.0
  * ImportExport_3DM 0.1.0
  * DynamicData 2.68.0
  * Pyramids-and-Polyhedrons
  * occi-freecad-plugin 0.1.0
  * GDML 2.0.0
  * fasteners 0.5.29
  * Design456 0.0.1
  * Alternate_OpenSCAD 1.0.0
  * SWPart
  * toSketch 1.0.1
  * Curves 0.6.50

luzpaz, 2024-10-16:
>  You might be loading two sets of Qt binaries into the same process. Check that all plugins are compiled against the right Qt binaries. Export DYLD_PRINT_LIBRARIES=1 and check that only one set of binaries are being loaded.

Any idea what's causing the dependency clash here ?

KeithSloan, 2024-10-18:
As well as test with FC Release Candidates I was hoping to try rhino3dm  8.0.1 or later see
https://github.com/jesterKing/import_3dm/issues/110

Could it be that the latest version of rhino3dm is a problem.

I tried uninstall, restart an install and got in report view
12:00:07  pip timed out
12:00:21  Attempting to fetch remote copy of ALLOWED_PYTHON_PACKAGES.txt...
12:00:21  Overriding local ALLOWED_PYTHON_PACKAGES.txt with newer remote version

Then this popup
[image]

No idea why it is looking in 0.22 for pip, other than I have this and a number of other versions on my machine.