FreeCAD/FreeCAD #17271: Measurement: Center of Mass mode shows 0 coordinates with minus

FEA-eng, 2024-10-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Maybe it's also the case with other modes but I've only noticed that for CoM:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-10-16:
The minus zero values you see are in fact not zero, they have been rounded to that value. We see this in other places as well.

FEA-eng, 2024-10-16:
> The minus zero values you see are in fact not zero, they have been rounded to that value. We see this in other places as well.

That's what I suspected but I expected the tool to remove the minus sign when rounding to 0,00.

Then maybe the issue should be more general?