FreeCAD/FreeCAD #17272: FreeCAD crashes when adding a point to arc in sketcher

tomunger, 2024-10-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD crashes when adding a point to an arc.  

I want to add a point to the arc where it crosses the axis.  

- Edit the sketch
- Add a point to the arc where it crosses the axis.  (Or anywhere on the arc, same problems)
- "Close" the sketch
- FreeCAD crashes.  On a Mac, I get the crash report.

If this action is allowed - and I think it is - freeCAD should not crash.  If it is not allowed, there should be an error message.

the attached ZIP file contains the FreeCAD file and the Macintosh crash report.

### Full version info

```shell
OS: macOS 14.7
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Manipulator 1.5.0
  * fasteners 0.5.21
  * lattice2 1.0.0
  * CurvedShapes 1.0.4
  * Curves 0.6.13
  * Assembly3 0.12.0
```

### Subproject(s) affected?

Sketcher

### Anything else?

[Point Bug.zip](https://github.com/user-attachments/files/17384314/Point.Bug.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-10-15:
Mac tends to crash when Linux just get error messages.
I thought I've fixed all these in quick measure. I'll check tomorrow if I can fix so it doesn't crash.

hyarion, 2024-10-16:
I can reproduce this issue on 1.0RC2, but not on latest weekly.

We did a lot of fixes for quick measure since rc2 so I'm not surprised that the crash was found, but I think we can close the issue.

@tomunger you can use a weekly release as a workaround until next 1.0 release comes out.
1. download weekly builds here: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
2. install freecad in /Application
3. open a terminal window and enter `sudo xattr -r -d com.apple.quarantine /Applications/FreeCAD.app` to make them work (there's currently an issue with signing preventing them to work as is)

tomunger, 2024-10-16:
Thanks.  It was not clear to me that weekly builds were for the 1.0 release.  I thought it possible that they were new development for the next release.  I'll switch to those. 
I confirmed that this operation works in the latest build.  I'm closing this issue.

closed 2024-10-16 by tomunger

hyarion, 2024-10-16:
@tomunger you are correct that weekly builds are for the next release, but issues like crashes are back ported to 1.0.