FreeCAD/FreeCAD #17280: MacOS/Sequoia/ARM  Immediate segfault after double-clicking document

dmopalmer, 2024-10-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When FreeCAD is not running, double clicking on a `.FCStd` document starts up the app, but it immediately segfaults.

When the app is started separately, and then the document is double clicked (or opened from the `File` menu) it does not crash.

Crash report stack dump and example file attached.
[einstein_fc.FCStd.zip](https://github.com/user-attachments/files/17388893/einstein_fc.FCStd.zip)
[freecad_crash_report.txt](https://github.com/user-attachments/files/17388895/freecad_crash_report.txt)

### Full version info

```shell
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * freecad.gears 1.2.0
  * 3D_Printing_Tools

macOs Sequoia 15.0.1 on Apple M1 16 GB Ram.  (More details in crash dump)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-10-16:
This is a known regression caused by the previous attempt to close the splash screen before the recompute-dialog: https://github.com/FreeCAD/FreeCAD/pull/17178

This is only in the main branch (weekly builds), the code causing this has not been merged to 1.0.

maxwxyz, 2024-10-17:
@chennes fyi

sliptonic, 2024-10-19:
Discussed at developer meeting.  Qualifies for $500 FPA bounty

JohnOCFII, 2024-10-19:
I've tested this with a few different files.  I am NOT seeing a crash with most files, but with the example file attached above, I AM seeing an issue.  macOS is calling that file "unknown or possibly malware."  Not sure if this remains a general issue or not.
[image]

This attached file, for example, opens successfully when double-clicking from Finder even when FreeCAD is not running.
[SeptemberTestBox.FCStd.zip](https://github.com/user-attachments/files/17446371/SeptemberTestBox.FCStd.zip)

```
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.50
```

oursland, 2024-10-19:
@JohnOCFII  That error is because the weekly builds are not signed.  To address it you will need to run `xattr -cr /Applications/FreeCAD.app` to remove the "downloaded from the internet" attribute.

maxwxyz, 2024-10-19:
@oursland so it shouldn't happen with the release candidates or are they also not signed? Issue was created with a RC version.

oursland, 2024-10-19:
I misread the error.  It appears that the downloaded file itself is being considered risky?

That's new to me and I'm unaware of how to address it.

hyarion, 2024-10-19:
> I've tested this with a few different files. I am NOT seeing a crash with most files, but with the example file attached above, I AM seeing an issue. macOS is calling that file "unknown or possibly malware." Not sure if this remains a general issue or not. <img alt="Screenshot 2024-10-19 at 11 34 26 AM" width="372" src="https://private-user-images.githubusercontent.com/7844212/378105784-9b1f2479-f1c6-4700-96b6-d593e6b899ee.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkzNjg3MDIsIm5iZiI6MTcyOTM2ODQwMiwicGF0aCI6Ii83ODQ0MjEyLzM3ODEwNTc4NC05YjFmMjQ3OS1mMWM2LTQ3MDAtOTZiNi1kNTkzZTZiODk5ZWUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAxOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMTlUMjAwNjQyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDA2NmY2Y2ZhNWU5YWJlNzUzZTVkMzFjMzBiNmMzMzkxMjk3ZGQzZjBmM2RkY2ZlOWQxYWMyNGU2Mjg4MmQwNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.HPzKrWSYx0elJvHzqqy79PsWinVOM_mlBFfe4PIMFrk">
> 
> This attached file, for example, opens successfully when double-clicking from Finder even when FreeCAD is not running. [SeptemberTestBox.FCStd.zip](https://github.com/user-attachments/files/17446371/SeptemberTestBox.FCStd.zip)
> 
> ```
> OS: macOS 15.0.1
> Word size of FreeCAD: 64-bit
> Version: 1.1.0dev.38991 (Git)
> Build type: Release
> Branch: main
> Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
> Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
> Locale: C/Default (C)
> Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
> Installed mods: 
>   * FeedsAndSpeeds 0.5.0
>   * OpenTheme 2024.9.1
>   * fasteners 0.5.29
>   * Curves 0.6.50
> ```

This is a new issue

dmopalmer, 2024-10-19:
Here is another, simpler, file that I made which does the same crash.

This was created from scratch competely within version 1.0.0rc2 .  I unzipped it and rezipped it (command line Zip 3.0, still on the same Mac) to check for corruption at that level and the rezipped file also crashes it.  (The attached file is of course zipped beyond the standard .FCStd zipped format, so double-zipped.)

[shroud_clip.FCStd.zip](https://github.com/user-attachments/files/17447162/shroud_clip.FCStd.zip)

dmopalmer, 2024-10-19:
I downloaded SeptemberTestBox and tried it on my Mac.  It, too, crashes FreeCAD on startup, but can be opened when FreeCAD is already running.

In other clues: I have only been using FreeCAD for a month or so, and have found it very crashy and beta.  I reported this bug because it was highly replicable and not previously reported as I could find.  But, for example:
- Starting up I always get messages like: `14:16:13  (qt.qpa.fonts) Populating font family aliases took 53 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost.`
- When I try to apply a thickness
  -  I tried just now on SeptemberTestBox, and it says 'Empty thickness created!'
  - Last week I tried it on another object and it couldn't find a library.  (I searched on the library name and found a previous Forum discussion without resolution.)
- Spontaneous crashes and object corruption are common enough that I tend to write python in a text editor and copy-paste it into the console to make things, rather than repeatedly trying to make the same thing interactively and getting a few steps in before I have to start all over.

Is that standard for FreeCAD (I see enough bug reports and Youtube tutorials saying things like 'at this point you have to repeat your work' to get the impression that it is).  Or is this particular bug something that only shows up on my Mac because, e.g. it is an M1 with Sequoia 15.0.1 with an initial Freecad installation with 0.21.1 that caused the hidden config files to set some internal flag to `crashy`?

JohnOCFII, 2024-10-19:
> Here is another, simpler, file that I made which does the same crash.

Yes, I get the same **potential malware** message with that file as well.  Not sure why.  

I've not seen that message before.

[image]

dmopalmer, 2024-10-19:
If you unzip it and re-zip it from the command line (so the file was 'created' on your own mac) does the message go away?

hyarion, 2024-10-19:
Please create a new issue for that regression

JohnOCFII, 2024-10-19:
> Or is this particular bug something that only shows up on my Mac because, e.g. it is an M1 with Sequoia 15.0.1 with an initial Freecad installation with 0.21.1 that caused the hidden config files to set some internal flag to `crashy`?

I too run an M1 MacBook Pro.  I'm not having the file issues you are having, but I can confirm that the ` (qt.qpa.fonts)` message is a known issue.  Also, sadly, `thickness` is known to be pretty likely to fail too.

You do mention that you had an older version (0.21.1) of FreeCAD on this Mac.  Can I suggest you either delete or rename the `user.cfg` and `system.cfg` files in the /Users/<USERNAME>/Library/Preferences/FreeCAD/  directory?  I'm wondering if starting with clean configuration files might resolve your issue.

JohnOCFII, 2024-10-19:
> If you unzip it and re-zip it from the command line (so the file was 'created' on your own mac) does the message go away?

I just tried that.  The "Not Opened" message still appears when I attempt to open from the Finder.  That file does open properly if I launch FreeCAD first.

[image]

JohnOCFII, 2024-10-19:
> This is a new issue

I'm not sure what you mean.  I'm just testing the OPs issue, and seeing the same results they are seeing, but only on files they've created.

hyarion, 2024-10-19:
@JohnOCFII 
It seems there's two bugs:
1. Loading freecad with an old file crashes that needs to be recomputed crashes RC2
2. Opening the downloaded file causes macOS to open a malware warning (I haven't seen that before) <-- this one is what I think we should create a new issue for :)

hyarion, 2024-10-19:
I just got an idea on how to debug this issue without waiting for a new weekly. I'll give this issue another try tomorrow.

dmopalmer, 2024-10-19:
I just created a new test account on my Mac (therefore no legacy config files) and tried opening the shroud_clip file.  I got the same crash.

Likewise, even in the new account it crashes on SeptemberTestBox, even though @JohnOCFII says it doesn't crash for him.

The stack trace (in the original post) might be informative 
```
Thread 0 Crashed:
Thread 0 Crashed:
0   libsystem_pthread.dylib       	       0x1916818e0 0x191680000 + 6368
1   libCoin.80.0.3.dylib          	       0x109eebd58 SoGLCacheContextElement::getUniqueCacheContext() + 28
2   libFreeCADGui.dylib           	       0x1076a05b0 SIM::Coin3D::Quarter::QuarterWidgetP::findCacheContext(SIM::Coin3D::Quarter::QuarterWidget*, QOpenGLWidget const*) + 184
3   libFreeCADGui.dylib           	       0x1076a04ac SIM::Coin3D::Quarter::QuarterWidgetP::QuarterWidgetP(SIM::Coin3D::Quarter::QuarterWidget*, QOpenGLWidget const*) + 156
4   libFreeCADGui.dylib           	       0x10769d078 SIM::Coin3D::Quarter::QuarterWidget::constructor(QSurfaceFormat const&, QOpenGLWidget const*) + 212
5   libFreeCADGui.dylib           	       0x10769d3c0 SIM::Coin3D::Quarter::QuarterWidget::QuarterWidget(QWidget*, QOpenGLWidget const*, QFlags<Qt::WindowType>) + 92
6   libFreeCADGui.dylib           	       0x1076a3b08 SIM::Coin3D::Quarter::SoQTQuarterAdaptor::SoQTQuarterAdaptor(QWidget*, QOpenGLWidget const*, QFlags<Qt::WindowType>) + 52
7   libFreeCADGui.dylib           	       0x1076cc390 Gui::View3DInventorViewer::View3DInventorViewer(QWidget*, QOpenGLWidget const*) + 44
8   libFreeCADGui.dylib           	       0x1076c7610 Gui::View3DInventor::View3DInventor(Gui::Document*, QWidget*, QOpenGLWidget const*, QFlags<Qt::WindowType>) + 252
9   libFreeCADGui.dylib           	       0x1072d1858 Gui::Document::createView(Base::Type const&) + 256
10  libFreeCADGui.dylib           	       0x10724e0c4 Gui::Application::slotNewDocument(App::Document const&, bool) + 2540
...etc...
```

JohnOCFII, 2024-10-20:
> I just created a new test account on my Mac (therefore no legacy config files) and tried opening the shroud_clip file. I got the same crash.
> 
> Likewise, even in the new account it crashes on SeptemberTestBox, even though @JohnOCFII says it doesn't crash for him.

Interesting.

Is there any system wide 3rd party software on this Mac?  Something that might impact file operations?  Something like Default Folder X, or the Rewind AI software?  Just curios, as I've not seen this particular issue before.

JohnOCFII, 2024-10-20:
> @JohnOCFII It seems there's two bugs:
> 
> 1. Loading freecad with an old file crashes that needs to be recomputed crashes RC2
> 2. Opening the downloaded file causes macOS to open a malware warning (I haven't seen that before) <-- this one is what I think we should create a new issue for :)

Thank you for clarifying.  While I'd heard of issue #1, I felt this thread was only dealing with issue #2, thus, if we were to create a separate thread, it would be for issue #1.

depthoffocus, 2024-10-20:
Could this issue be to do with file associations? Something new to Sequoia? 

Looks like it's not unique to FreeCAD. Seemingly similar issues in Apple Discussions:

https://discussions.apple.com/thread/255787115

https://discussions.apple.com/thread/255803194

https://discussions.apple.com/thread/255762766

JohnOCFII, 2024-10-20:
> Could this second issue of the malware warning be to do with file associations? Something new to Sequoia?

I think you are right!  I'll have to try this app on my older non-Sequoia Mac, but I did follow the suggestion in one of those threads to select the file in question, and select the option to "open all files of this type with this App.

It really DOES seem to be a macOS Finder bug related to their ever-increasing "security." 

Here are the screen shots of the steps I followed to get this file to open via Double-Click in Finder:

Select and Right-Click on the file in question, and choose `Get Info`

[image]

Then choose the option to option to `Change All` to launch with the desired version of FreeCAD.
[image]

[image]

After following this chain of thought, I also looked into the Security settings on the Mac, and after you receive the initial lockout notice, you have the option to launch the App from within the security settings too.

So -- this issue is really a macOS issue!

Workaround is as above.  Now to test on older macOS, where I am guessing this won't be an issue.

JohnOCFII, 2024-10-20:
And yes, @hyarion -- after all that -- if you DO attempt to open an older file that needs to be recomputed, it WILL cause a crash. ;)

depthoffocus, 2024-10-20:
> Workaround is as above. Now to test on older macOS, where I am guessing this won't be an issue.

I am not yet on Sequoia and I haven't been able to reproduce this issue.

JohnOCFII, 2024-10-20:
> this is a separate issue, right? So this issue cannot be fixed currently, correct?

Two intermingled issues in this thread.  One is a FreeCAD issue.  The other appears to be a MacOS issue only impacting the most recent macOS release, Sequoia, (macOS 15).  Apple increased the level of malware protection, making it more difficult to deal with unknown apps.  (This is one reason I believe all FreeCAD releases, including weeklies, should be signed going forward -- as you will no longer be able to just "right click and allow" access).  But the particular problem here is that it seems to be treating unknown file types (in this case, FCStd files) as `Apps` and since document files shouldn't need to be signed, but it doesn't recognize the file -- we are stuck.  Somehow, using the option to tell macOS to open all files of this type with a version of FreeCAD seems to get around the issue.

Here's Apple's write-up on what they expect to happen.  Notice that there is no mention of document files being part of this situation: https://support.apple.com/en-ie/102445

depthoffocus, 2024-10-20:
I can only assume that Apple will have to modify this behaviour going forward because it appears to be affecting  really common office formats like Word and CSV files, and it's wholly unexpected.

But I also hope Mac FreeCAD builds will be signed going forward.

JohnOCFII, 2024-10-20:
> > Workaround is as above. Now to test on older macOS, where I am guessing this won't be an issue.
> 
> I am not yet on Sequoia and I haven't been able to reproduce this issue.

I just tested on Sonoma (on a different M1 MacBook Pro) and did not see this "file could not be checked for malware"  issue.  It is macOS Sequoia (15) specific.

maxwxyz, 2024-10-20:
@chennes can we remove the blocker label and bounty? It only affects one OS and only one version and it seems something which Apple needs to fix,

chennes, 2024-10-20:
Are we still talking about the segfault on double-click? I don't see how the malware thing is related (and our release will be signed).

JohnOCFII, 2024-10-20:
> @chennes can we remove the blocker label and bounty? It only affects one OS and only one version and it seems something which Apple needs to fix,

Based on the call yesterday, the bounty is related to the _other_ topic which is wrapped up in this issue which is that ALL older FreeCAD files that need to be recomputed WILL cause a segfault on macOS.  Not sure if there is already a separate ticket for that issue, or if this issue needs to be split.

JohnOCFII, 2024-10-20:
FYI -- the workaround in this other item (https://github.com/FreeCAD/FreeCAD/pull/17367) is the one that should get the bounty.

chennes, 2024-10-20:
Yes, agreed.

closed 2024-10-21 by yorikvanhavre

hyarion, 2024-10-22:
I've tested the new build, with the file attached to this issue, and it works for me.
@dmopalmer @JohnOCFII can you test the new weekly and see if it works for you too?

JohnOCFII, 2024-10-22:
> I've tested the new build, with the file attached to this issue, and it works for me.
> @dmopalmer @JohnOCFII can you test the new weekly and see if it works for you too?

I tested by double-clicking from the macOS Finder with a freshly unzipped version of the attached einstein_fc.FCStd file, and it prompted for recomputation.  The program did not crash after recomputation.  I'd say that issue has been resolved.

```
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.50
```

maxwxyz, 2024-10-23:
@dmopalmer could you test too: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

dmopalmer, 2024-10-24:
Something is wrong with the build.

On my Mac `FreeCAD_weekly-builds-39024-conda-macOS-arm64-py311.dmg` fails to open:.  The `verifying` step says
"“FreeCAD.app” is damaged and can’t be opened. You should eject the disk image."

The SHA256sum of the downloaded  `.dmg` matches the value on github.

oursland, 2024-10-24:
The weekly builds are unsigned.  To run the build you must run `xattr -c /Applications/FreeCAD.app` to remove the quarantine attribute that is added to files downloaded from the internet.

dmopalmer, 2024-10-24:
After xattr, the build successfully starts up when I double click an `.FCStd` file.  (Both the shroud and clip files I supplied ).

So success.

chennes, 2024-11-06:
Noting that @hyarion is eligible to claim the bounty from the FPA. To claim, please send an email to fpa@freecad.org referencing this issue, and your preferred method of payment.

kknight, 2024-11-14:
Had the same problem but the problem turned out to be that the RC3 build requires python3.9 on the system, the solution was:

`brew install python@3.9`

hyarion, 2024-11-14:
> Had the same problem but the problem turned out to be that the RC3 build requires python3.9 on the system, the solution was:
> 
> `brew install python@3.9`

Whatever happened there seems to be fixed in RC4.

Looking at open files it successfully uses the bundled python version:
```
/Applications/FreeCAD.app/Contents/Resources/lib/libpython3.11.dylib
/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/lib-dynload/zlib.cpython-311-darwin.so
etc.
```

PR #17367 (2024-10-21): Workaround for macOS crash when loading FreeCAD by opening a `.FCStd` file from Finder
Fixes #17280

TL;DR
Files are loaded before main window is loaded, things go kaboom.
Delay the load event a bit fixes the issue

Please, don't squash this PR, one of the commit reverts an earlier attempt, it makes sense to keep that revert that change in a separate commit.
Only backport the second commit "Repost QFileOpenEvent event to avoid an undefined state", as the previous attempt was never backported.

---

When the user double clicks a `.FCStd` file in Finder when FreeCAD isn't open, FreeCAD will star [... cut]