FreeCAD/FreeCAD #17281: additive loft not working in selecting sketch

apngpt, 2024-10-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While selecting the various cross-sectional area for the loft feature, it is not working. 

### Full version info

```shell
RC2
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

apngpt, 2024-10-17:
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/India (en_IN)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.63
  * Assembly4 0.50.12
  * Manipulator 1.5.7
  * sheetmetal 0.4.11
[image]
[image]

luzpaz, 2024-10-17:
@apngpt you are using an inferior version of sheetmetal workbench (https://github.com/shaise/FreeCAD_SheetMetal?tab=readme-ov-file#release-notes) which is currently at v0.4.26  
Can you update and see if that changes anything ? (you may need to uninstall and re-install it since @shaise's workbenches have an issue with the addon manager

reegeenee, 2024-10-21:
_OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion_

@luzpaz I got the same error with complete new installed Version.
The error occurs, if you  select 1st the sketch, select then additive loft function and add there the other sketches.
If you select all sketches directly in the tree and use then additive loft function, it works well.

reegeenee, 2024-10-22:
here is an example
[cover.zip](https://github.com/user-attachments/files/17475372/cover.zip)

ikemedes, 2024-11-05:
Same here:
Version 1.0.0RC2 Rev. 38806 New and clean installation.
OS Ubuntu 24.04.1 LTS New and clean installation.
Part Design - Additive Loft only works when first selecting sketches from the tree, then calling (selecting) the additive loft modifier/tool. 
This is contrary to the information in the wiki.
But selecting a **face** in 3d view then selecting additive loft modifier/tool, then from that select a sketch, it works.
This works as per wiki.

reegeenee, 2024-11-21:
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1

In new Version it is now working correctly.

FEA-eng, 2025-09-14:
I can't reproduce it atm.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/09 06:43:51
Build type: Release
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1

```