FreeCAD/FreeCAD #17284: Feature: BIM_leader Filled and Close options not applicable for the tool

semhustej, 2024-10-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Priority: very low

'Filled' option does not seem make sense for BIM_leader command in my opinion. 
[image]

'Close' option does not seem make sense for BIM_leader command in my opinion. 
[image]

Both of these options could be removed from the tool in my opinion, so that they won't confuse users.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 (Git)
Build type: Release
Branch: main
Hash: e478192aa5393de9b7df190341a043f79ec6ac49
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-16:
@paullee0 can you opine on this ? TIA

yorikvanhavre, 2024-10-17:
Indeed, makes sense. This would be relatively easy to implement, one needs to add one more initializing option at https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/BIM/bimcommands/BimLeader.py#L38 that hides these two controls

PR #17310 (2024-11-16): BIM: Optimized leader UI
Fixes #17284
commit 89d1d717 (2024-11-16): BIM: Optimized leader UI (#17310)

* BIM: Optimized leader UI - fixes #17284

* Correct task panel icon for BIM Leader.
* Make BIM Leader task panel title translatable.
* Use a single mode parameter for the Line class.

---------
closed 2024-11-16 by Roy-043