FreeCAD/FreeCAD #17286: Part: Extra geometry in a boolean operation on a cone.

kaiwas, 2024-10-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As a result of the operation of subtracting a cone from an object, extra geometry is created.
At the same time, in the case of subtracting a sphere, such a problem is not observed.

[image]
[image]

The file for tests is attached. Remove the zip extension.
[Cone_Problem.FCStd.zip](https://github.com/user-attachments/files/17392818/Cone_Problem.FCStd.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git) AppImage
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * FreeCAD_SketchArch
  * lattice2 1.0.0
  * PitchedRoof
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-10-16:
In the German sub-forum the same problem was reported: https://forum.freecad.org/viewtopic.php?t=91426

kaiwas, 2024-10-16:
> Here is the culprit: #15931

Thanks for the info.
It looks like these changes are already included in the weekly build, but for some reason the problem still exists.

wwmayer, 2024-10-16:
> but for some reason the problem still exists.

These changes are causing the problem. For builds with OCC 7.7.2 not a full cone is created but one with a gap. When using the cone for a boolean cut then it leaves this extra face you see.

I have a FreeCAD build with OCC 7.7.2 here and I cannot see that this workaround is needed at all.

closed 2025-08-15 by kaiwas