FreeCAD/FreeCAD #17292: Measurement: Saved measurement always visible after reload

wwmayer, 2024-10-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://forum.freecad.org/viewtopic.php?t=91435

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (XFCE/xubuntu/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38979 (Git)
Build type: debug
Branch: main
Hash: 05111333272c420bc999052a12de040fdd919413
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/Classic theme/Qt default
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-16:
cc @hlorus

luzpaz, 2024-10-16:
Per wwmayer in the discussion:

> The problem is that the visibility of the measurement object depends on the visibility of the source object.
>1. Hide the measurement
>2. Hide the body
>3. Show the body
>Result: the measurement is visible
>
> 1. Hide the body
Result: the measurement is invisible. And it's not even possible to show the measurement object
>
> The function to blame is `ViewProviderMeasureBase::onSubjectVisibilityChanged`

hlorus, 2024-10-21:
The described issue (visible after reload) seems to be the intended behaiviour according to this comment: https://github.com/FreeCAD/FreeCAD/blob/ef2156d0f6d116b1642a5531a2d44fc6b7d17042/src/Mod/Measure/Gui/ViewProviderMeasureBase.cpp#L232. Honestly i don't remember what the reasoning was behind that, maybe @WandererFan remembers. Anyway the behaiviour can easily be changed by removing the call to `show()`.

The issue described by @wwmayer (visibility depends on source object) is also intended behaiviour as otherwise there are context-free measurements floating in space, whenever a source object is hidden, which is annoying. I agree however that it should be possible to unhide a measurement manually which doesn't seem to be possible currently.

There is one other issue related to visibility that i came across: When unhiding a source object it's measurements are always visible again, even if the measurement was manually hidden by the user. This is because the hiding/unhiding the source object uses the same visibility property as "manual" hiding/unhiding. Should there be two distinct properties?

WandererFan, 2025-08-15:
I'm perfectly ok with a measurement not being visible if the subject is hidden.  

truth table for measurement visibility 
```
subject visible                t   t   f   f
measurement visible     t   f   t   f
show measurement       t   f   f   f
```
Needs tweaks to onSubjectVisibilityChanged and finishRestoring to implement this true table.