FreeCAD/FreeCAD #17294: Measurement: Distance Measurement between parallel faces does not work between two inidividual bodies

LangerJanD, 2024-10-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Measuring the Distance between parallel faces of one body works fine. When doing the same measurement between (parallel) faces of two individual bodies a tangent dimension is returned.
[image]

I get the same result for Auto and Distance measurement modes:

    Click on Caliper. Mode is Auto. Click on face A then click on face B
    Click on Caliper. Click on reset, select Mode Distance. Click on face A then click on face B

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus 0.4.61
  * Assembly3 0.12.0
  * Assembly4 0.50.4
  * Defeaturing 1.2.1
  * fasteners 0.4.70
  * fasteners.backup1696421279.5348685 0.4.62 (Disabled)
  * Help 1.0.3
  * kicadStepUpMod 10.21.3
  * kicadStepUpMod.backup1697201420.3722107 10.19.5 (Disabled)
  * Manipulator 1.5.0
  * sheetmetal 0.3.12
  * Woodworking 0.21.30486
```

### Subproject(s) affected?

None

### Anything else?

[MeasurementBug.zip](https://github.com/user-attachments/files/17395646/MeasurementBug.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-16:
Can you please test on RC2, @LangerJanD ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2)

LangerJanD, 2024-10-16:
With 1.0RC2 the the measurement still returns the wrong value: 
[image]
When the newly added checkbox Show Delta is enabled I see the correct result as Delta X. But this would still fail if the face is not parallel to any axis:
[image]

Version info:
------------------------------
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Dark/Qt default
Installed mods: 
  * A2plus 0.4.61
  * Assembly3 0.12.0
  * Assembly4 0.50.4
  * Defeaturing 1.2.1
  * fasteners 0.4.70
  * fasteners.backup1696421279.5348685 0.4.62 (Disabled)
  * Help 1.0.3
  * kicadStepUpMod 10.21.3
  * kicadStepUpMod.backup1697201420.3722107 10.19.5 (Disabled)
  * Manipulator 1.5.0
  * sheetmetal 0.3.12
  * Woodworking 0.21.30486

luzpaz, 2024-10-16:
[image]

```
OS: Manjaro Linux (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git) AppImage
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

kaiwas, 2024-10-16:
> With 1.0RC2 the the measurement still returns the wrong value:

why is the value wrong?
35*35+15*15=38.08*38.08=1450

luzpaz, 2024-10-19:
@hlorus can you take a look at this?

luzpaz, 2025-08-27:
@hlorus any chance you can take a look before 1.1 release ?

luzpaz, 2026-04-29:
ticket renamed based on suggestion in https://github.com/FreeCAD/FreeCAD/issues/17294#issuecomment-4331094591

FEA-eng, 2026-11-07:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.14/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```