FreeCAD/FreeCAD #17297: Draft/BIM: unit system is reset to default unit when switching between unsaved documents

semhustej, 2024-10-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Priority: low

When switching between **unsaved** documents, units setup (which have been set-up in the units switcher in the bottom right corner) switch back to default, when switching between **unsaved** documents. When switching between saved documents, unit system is preserved for the document.

Steps to reproduce:

- Open FreeCAD
- Open new document 01 and set units to US customary
- Open another new document 02 and set units to Imperial Civil
- Switch back to document 01. The units are set up to default settings.
- Open the units set-up in the bottom right corner -> US customary is still selected, but not valid

See the behavior in the video:

https://github.com/user-attachments/assets/fb021106-3e7d-441a-97ba-9a7710cee07d

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 (Git)
Build type: Release
Branch: main
Hash: e478192aa5393de9b7df190341a043f79ec6ac49
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Draft

### Anything else?

The units system seem to behave in a different way in 0.21.2, so I was not able to conclude the correct behavior. The above described looks like a bug though.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-10-17:
This is a Core issue. The same test can be done with the [Part_Primitives](https://wiki.freecad.org/Part_Primitives) command for example. IMO this is a blocker.

sliptonic, 2024-10-19:
Next step might be a bisect to find out exactly when it was introduced

wwmayer, 2024-10-20:
https://github.com/FreeCAD/FreeCAD/commit/a1c08dbf43d4f887a1ac4682360264cb390f83ad#diff-10e12047335caea2a9bcf956215f783d19e1e2163560e23f91ce16151972d48dR940

PR #17366 (2024-10-21): Core: Set unit scheme when switching between unsaved documents
Fixes #17297
commit 5435b3e4 (2024-10-21): Core: Set unit scheme when switching between unsaved documents

Fixes #17297
commit 109f481a (2024-10-21): Core: Set unit scheme when switching between unsaved documents

Fixes #17297
closed 2024-10-21 by yorikvanhavre