FreeCAD/FreeCAD #17300: PartDesign: Crash Creating a Datum Plane

point-droid, 2024-10-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD 1.0RC2 crashes when trying to create a Datum Plane using the X, Y or Z axis
1. Open FreeCAD RC2
2. File New
3. Switch to PartDesign
4. Create Body
5. Click PartDesign_Plane
6. Switch to Model tab
7. Expand Origin
8. Select Either X-axis or Y-axis or Z-axis

0.21.2 will allow you to do this so I guess this is a regression.

`Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x4adc0) [0x7f7d86f45dc0]
#1  0x7f7d74118fc6 in BRep_Tool::Curve(TopoDS_Edge const&, TopLoc_Location&, double&, double&) from /tmp/.mount_FreeCAsgdCIv/usr/lib/./libTKBRep.so.7.7+0x26
#2  0x7f7d7412ab21 in BRepAdaptor_Curve::Initialize(TopoDS_Edge const&) from /tmp/.mount_FreeCAsgdCIv/usr/lib/./libTKBRep.so.7.7+0xd1
#3  0x7f7d7412bd35 in BRepAdaptor_Curve::BRepAdaptor_Curve(TopoDS_Edge const&) from /tmp/.mount_FreeCAsgdCIv/usr/lib/./libTKBRep.so.7.7+0xd5
#4  0x7f7d7420fdce in Measure::Measurement::length() const from /tmp/.mount_FreeCAsgdCIv/usr/lib/Measure.so+0x15e
#5  0x7f7d5bfd0de7 in MeasureGui::QuickMeasure::printResult() from /tmp/.mount_FreeCAsgdCIv/usr/lib/MeasureGui.so+0x947
#6  0x7f7d5bfd19ee in MeasureGui::QuickMeasure::processSelection() from /tmp/.mount_FreeCAsgdCIv/usr/lib/MeasureGui.so+0x1e
#7  /tmp/.mount_FreeCAsgdCIv/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7f7d875c06f2]
#8  0x7f7d875c434a in QTimer::timeout(QTimer::QPrivateSignal) from /tmp/.mount_FreeCAsgdCIv/usr/bin/../lib/libQt5Core.so.5+0x2a
#9  0x7f7d875b564d in QObject::event(QEvent*) from /tmp/.mount_FreeCAsgdCIv/usr/bin/../lib/libQt5Core.so.5+0x7d
#10  0x7f7d880b63ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAsgdCIv/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#11  0x7f7d899ef079 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAsgdCIv/usr/bin/../lib/libFreeCADGui.so+0x99
#12  0x7f7d8758bb28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAsgdCIv/usr/bin/../lib/libQt5Core.so.5+0x108
#13  0x7f7d875def51 in QTimerInfoList::activateTimers() from /tmp/.mount_FreeCAsgdCIv/usr/bin/../lib/libQt5Core.so.5+0x381
#14  /tmp/.mount_FreeCAsgdCIv/usr/bin/../lib/libQt5Core.so.5(+0x30a75c) [0x7f7d875df75c]
#15  /tmp/.mount_FreeCAsgdCIv/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7f7d83649ad7]
#16  /tmp/.mount_FreeCAsgdCIv/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7f7d8364cac7]
#17  /tmp/.mount_FreeCAsgdCIv/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f7d8364d0ce]
#18  0x7f7d875dfa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAsgdCIv/usr/bin/../lib/libQt5Core.so.5+0x66
#19  0x7f7d8758a70b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAsgdCIv/usr/bin/../lib/libQt5Core.so.5+0x11b
#20  0x7f7d87592212 in QCoreApplication::exec() from /tmp/.mount_FreeCAsgdCIv/usr/bin/../lib/libQt5Core.so.5+0x82
#21  0x7f7d8994e4e1 in Gui::Application::runApplication() from /tmp/.mount_FreeCAsgdCIv/usr/bin/../lib/libFreeCADGui.so+0xa41
#22  /tmp/.mount_FreeCAsgdCIv/usr/bin/freecad(+0x789f) [0x558ffe71689f]
#23  /lib64/libc.so.6(__libc_start_main+0xef) [0x7f7d86f3024d]
#24  /tmp/.mount_FreeCAsgdCIv/usr/bin/freecad(+0x7c12) [0x558ffe716c12]
`

### Full version info

```shell
OS: openSUSE Leap 15.5 (KDE//usr/share/xsessions/plasma5)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * parts_library
  * CurvedShapes 1.0.10
  * Curves 0.6.48
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-16:
Cannot confirm on Windows, all working. Could you try this in the latest dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

```
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.50
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)

```

wwmayer, 2024-10-16:
It was fixed with https://github.com/FreeCAD/FreeCAD/pull/16921

closed 2024-10-17 by maxwxyz