FreeCAD/FreeCAD #17303: Included Robot-Example fails to migrate

Samu152, 2024-10-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The included example file "RobotExample.FCStd" fails to migrate on multible newer versions.

Console output:
15:19:00  Sketch: This feature is deprecated and won't be longer supported in future FreeCAD versions
15:19:00  Bein1: No object linked
15:19:00  Bein2: No object linked
15:19:00  Bein3: No object linked
15:19:00  Bein4: No object linked
15:19:00  PartToWork: No object linked

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Austria (de_AT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Anything else?

https://forum.freecad.org/viewtopic.php?p=787404#p787404

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-10-17:
How about dropping the robot example file(s) for 1.0?
We can add it later if someone fixes the issues.

adrianinsaval, 2024-10-17:
> How about dropping the robot example file(s) for 1.0? We can add it later if someone fixes the issues.

yeah that's better than shipping a broken model IMO

PR #17383 (2024-10-22): Remove broken robot example file
Fixes #17303

Note: The example file won't be removed from your build directory when building incrementally.
commit 4badd251 (2024-10-22): Remove broken robot example file

fixes #17303
commit e6f97126 (2024-10-22): Remove broken robot example file

fixes #17303
closed 2024-10-22 by chennes