FreeCAD/FreeCAD #17307: [Techdraw][Version 1.1.0dev.38977] Cannot change Leader line to dash, dot, dash and dot, or no line style

kevinsmia1939, 2024-10-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When drawing line with leader line. The line style cannot be change from continuous to anything else.
Version 0.21.2 does not have this problem. Only the newest build.
Left: 0.21.2
Right: 1.1.0dev.38977

I attached example file. Remove .txt before opening.
[image]
[techdraw_dash_line.FCStd.txt](https://github.com/user-attachments/files/17409715/techdraw_dash_line.FCStd.txt)

### Full version info

```shell
OS: openSUSE Tumbleweed (KDE/plasma6/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git) AppImage
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * QuickMeasure 2022.10.28
  * Assembly4 0.50.15
  * A2plus 0.4.68
  * ExplodedAssembly
  * Manipulator 1.5.7
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-10-17:
Missed in the migration to ISO standard line patterns.

luzpaz, 2024-10-17:
Target for 1.0 ?

commit 81070b1f (2024-10-21): [TD]Fix leader line style (fix #17307)
commit 718c2746 (2024-10-21): [TD]Fix leader line style (fix #17307)
closed 2024-10-21 by yorikvanhavre