FreeCAD/FreeCAD #17308: FreeCAD crashes on touch input under Linux

plundell, 2024-10-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**Reproduction:**

1. Connect touchscreen to laptop and ensure it works.
2. Start FreeCAD.
3. Touch _anywhere_ inside the FreeCAD window and **FreeCAD crashes** with this error:
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f4812642520]
#1  0x7f480d0e08cf in QXcbConnection::xi2ProcessTouch(void*, QXcbWindow*) from /tmp/.mount_freeca6W6ahl/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5+0x2f
#2  0x7f480d0af5f3 in QXcbConnection::handleXcbEvent(xcb_generic_event_t*) from /tmp/.mount_freeca6W6ahl/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5+0x183
#3  0x7f480d0b0a96 in QXcbConnection::processXcbEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freeca6W6ahl/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5+0x66
#4  /tmp/.mount_freeca6W6ahl/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64be3) [0x7f480d0d5be3]
#5  /tmp/.mount_freeca6W6ahl/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7f4810f10ad7]
#6  /tmp/.mount_freeca6W6ahl/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7f4810f13ac7]
#7  /tmp/.mount_freeca6W6ahl/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f4810f140ce]
#8  0x7f4812d0aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freeca6W6ahl/usr/bin/../lib/libQt5Core.so.5+0x66
#9  0x7f4812cb570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freeca6W6ahl/usr/bin/../lib/libQt5Core.so.5+0x11b
#10  0x7f4812cbd212 in QCoreApplication::exec() from /tmp/.mount_freeca6W6ahl/usr/bin/../lib/libQt5Core.so.5+0x82
#11  0x7f48158c84e1 in Gui::Application::runApplication() from /tmp/.mount_freeca6W6ahl/usr/bin/../lib/libFreeCADGui.so+0xa41
#12  /tmp/.mount_freeca6W6ahl/usr/bin/freecad(+0x789f) [0x55852b2b289f]
#13  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f4812629d90]
#14  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f4812629e40]
#15  /tmp/.mount_freeca6W6ahl/usr/bin/freecad(+0x7c12) [0x55852b2b2c12]
```

### Full version info

```shell
OS: Linux Mint 20.1 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Manipulator 1.5.7
  * Help 1.0.3
  * Curves 0.6.48
  * A2plus 0.4.67
```

### Subproject(s) affected?

None

### Anything else?

Laptop: Lenovo ThinkPad T490s 20NX003UMH
OS: Linux Mint 20.1
Kernel: 5.18.19-051819-generic
Touchscreen: Lenovo ThinkVision m14t (USB C)

I have tried on main branch FreeCAD 0.19 and 0.20, as well as realthunder's link branch 0.21 with the same result.

I have opened a sister-issue on realthunder's link branch: https://github.com/realthunder/FreeCAD/issues/1038

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-19:
Anyone able to decipher the stacktrace?

hyarion, 2024-10-20:
The issue is deep inside of Qt, according to the stacktrace, it is in this function (or an older version of it): https://github.com/qt/qtbase/blob/e498a076ae035588ed6f29e91a20eb727d608342/src/plugins/platforms/xcb/qxcbconnection_xi2.cpp#L901

Something similar has been reported here and it looks like the touch screen doesn't play nice with qt: https://bugreports.qt.io/browse/QTBUG-97603

I don't have any more info to give than maybe try and update and see if that solves the issue?

luzpaz, 2024-10-21:
Thanks @hyarion for weighing-in

@plundell most likely to get a grasp on this issue we need some more of your help.
1. We need you to test a build of FreeCAD that is built against Qt6 (your's is built against Qt5)
2. This will require running an experimental bleeding edge build of FreeCAD to do so.
Would you be willing to do so ?

@adrianinsaval https://github.com/FreeCAD/FreeCAD-Bundle/issues/303 also is asking for a qt6 build. Is there an ETA on a Qt6 support for the AppImage ?

hyarion, 2024-10-21:
@luzpaz hopefully qt won't be upgraded before 1.0 is released. Upgrading qt before 1.0 is out the door would make testing regression much more difficult.

adrianinsaval, 2024-10-21:
qt6 will be used post 1.0, a decision was made to remain on qt5 for 1.0

adrianinsaval, 2024-10-21:
but you can try the flathub beta version that is built with qt6.
```
flatpak remote-add --if-not-exists flathub-beta https://flathub.org/beta-repo/flathub-beta.flatpakrepo
flatpak install flathub-beta org.freecad.FreeCAD

```

luzpaz, 2024-10-21:
@plundell can you test the flatpak beta ? :point_up:

adrianinsaval, 2024-10-21:
In the mentioned upstream issue it suggested that it's a duplicate of https://bugreports.qt.io/browse/QTBUG-123554, there is a fix mentioned there. I'll see about backporting that to qt5 in conda this week

plundell, 2024-10-22:
@luzpaz I've now downloaded an run the flatpak - partial success. FreeCAD no longer crashes and I can move the cursor with finger or pen on touchscreen, but I cannot click with finger or pen (with mouse works fine). There are no errors printed in  the terminal.

The flatpak's _About_ read:
OS: KDE Flatpak runtime (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 3d63fc6)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/

I wrote this macro to see if I could glenn anything else:
```
from PySide import QtCore, QtGui

# Define a custom event handler
class EventFilter(QtCore.QObject):
    def __init__(self):
        super(EventFilter, self).__init__()
        self.current_focus_widget = None  # Store the current focused widget

    def eventFilter(self, obj, event):
		# Check the current focus widget
        new_focus_widget = QtGui.QApplication.focusWidget()

        # Only print the new focus widget if it changes
        if new_focus_widget != self.current_focus_widget:
            self.current_focus_widget = new_focus_widget
            print(f"Focus changed to: {self.current_focus_widget}")
            print("")
	
        if event.type() in (QtCore.QEvent.Paint, QtCore.QEvent.UpdateRequest):
            return False  # Let the event continue without logging it
	
        print(event)
        print("")

        return super(EventFilter, self).eventFilter(obj, event)

FreeCADGui.getMainWindow().installEventFilter(EventFilter())
```
...and found the following:

- when moving the cursor around with the mouse I got a flood of these events:
```
<PySide6.QtGui.QMouseEvent(MouseMove pos=648,82 scn=648,82 gbl=648,108 dev=QInputDevice(QPointingDevice("Virtual core pointer" Mouse id=2 seat=30002 caps=Position|Scroll|Hover))>

<PySide6.QtGui.QHoverEvent(HoverMove pos=648,82 scn=648,82 gbl=648,108 dev=QInputDevice(QPointingDevice("Virtual core pointer" Mouse id=2 seat=30002 caps=Position|Scroll|Hover))>

<PySide6.QtGui.QMouseEvent(MouseMove pos=658,84 scn=658,84 gbl=658,110 dev=QInputDevice(QPointingDevice("Virtual core pointer" Mouse id=2 seat=30002 caps=Position|Scroll|Hover))>

<PySide6.QtGui.QHoverEvent(HoverMove pos=658,84 scn=658,84 gbl=658,110 dev=QInputDevice(QPointingDevice("Virtual core pointer" Mouse id=2 seat=30002 caps=Position|Scroll|Hover))>
```
- when I clicked with the mouse I got:
```
<PySide6.QtGui.QMouseEvent(MouseButtonPress LeftButton pos=392,255 scn=392,255 gbl=392,281 dev=QInputDevice(QPointingDevice("Virtual core pointer" Mouse id=2 seat=30002 caps=Position|Scroll|Hover))>
```
- when I touched the screen with my finger the cursor did indeed move with it, but i only got these 3 events no matter how long I moved it for
```
<PySide6.QtGui.QTabletEvent(TabletMove NoButton pos=684.229,273.224 z=0 xTilt=0 yTilt=0 pressure=0 dev=QInputDevice(QPointingDevice("Wacom Co.,Ltd. Pen and multitouch sensor Stylus stylus" Stylus id=15 seat=30002 ptrType=Pen caps=Position|Pressure|MouseEmulation|Hover|XTilt|YTilt uniqueId=0))>

<PySide6.QtCore.QEvent(QEvent::StatusTip)>

<PySide6.QtCore.QEvent(QEvent::CursorChange)>

<PySide6.QtGui.QHoverEvent(HoverMove pos=684,273 scn=684.229,273.224 gbl=2604,323 dev=QInputDevice(QPointingDevice("Virtual core pointer" Mouse id=2 seat=30002 caps=Position|Scroll|Hover))>
```
- and same deal with the stylus
```
<PySide6.QtGui.QTabletEvent(TabletMove NoButton pos=1122.17,333.794 z=0 xTilt=0 yTilt=0 pressure=0 dev=QInputDevice(QPointingDevice("Wacom Co.,Ltd. Pen and multitouch sensor Stylus stylus" Stylus id=15 seat=30002 ptrType=Pen caps=Position|Pressure|MouseEmulation|Hover|XTilt|YTilt uniqueId=0))>

<PySide6.QtGui.QHoverEvent(HoverMove pos=1122,334 scn=1122.17,333.794 gbl=3042,384 dev=QInputDevice(QPointingDevice("Virtual core pointer" Mouse id=2 seat=30002 caps=Position|Scroll|Hover))>

<PySide6.QtGui.QTabletEvent(TabletMove NoButton pos=1122.17,333.794 z=0 xTilt=0 yTilt=0 pressure=0 dev=QInputDevice(QPointingDevice("Wacom Co.,Ltd. Pen and multitouch sensor Stylus stylus" Stylus id=15 seat=30002 ptrType=Pen caps=Position|Pressure|MouseEmulation|Hover|XTilt|YTilt uniqueId=0))>
```
- and finally, I got nothing if I tried to "click" using my finger or the buttons on the stylus

I hope this helps. Let me know if you need anything more

luzpaz, 2024-10-23:
@plundell you're a legend! Thank you for taking the time. This is a great place for us to start and diagnose more specific issues. Please stay subscribed to this ticket so if we need more testing, TIA!

adrianinsaval, 2024-10-24:
I will attempt back porting the fix to qt5

luzpaz, 2025-08-29:
@plundell is this still an issue?

plundell, 2025-08-29:
I will check and report back

plundell, 2025-09-08:
Ok, now I'm using AppImage version 1.0.2 revision 39319 (Git). 

It no longer crashes when I touch it with the pen, hurray! 

However (and this should probably be a new ticket), most of the time when I try to launch it it crashes immediately with the below output in the terminal. If I keep trying to launch it eventually (2-6 tries) works. 

```
pl-laptop-02 ~ $ /opt/freecad/FreeCAD_1.0.2-conda-Linux-x86_64-py311.AppImage 
FreeCAD 1.0.2, Libs: 1.0.2R39319 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f3b42c42520]
#1  /lib/x86_64-linux-gnu/libc.so.6(free+0x1e) [0x7f3b42ca53fe]
#2  /tmp/.mount_FreeCAKbaCiN/usr/lib/python3.11/lib-dynload/readline.cpython-311-x86_64-linux-gnu.so(PyInit_readline+0x15e) [0x7f3b37f8f68e]
#3  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x2cb39c) [0x7f3b452cb39c]
#4  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x201e2b) [0x7f3b45201e2b]
#5  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7f3b451ecfd8]
#6  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f3b4520b3af]
#7  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x1faba6) [0x7f3b451faba6]
#8  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7f3b452299df]
#9  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7f3b452291f7]
#10  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x23e3be) [0x7f3b4523e3be]
#11  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3546) [0x7f3b451f3546]
#12  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7f3b451ecfd8]
#13  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f3b4520b3af]
#14  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5704) [0x7f3b451d5704]
#15  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x1035dd) [0x7f3b451035dd]
#16  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x29ef1d) [0x7f3b4529ef1d]
#17  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7f3b4529e914]
#18  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x2b54be) [0x7f3b452b54be]
#19  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3546) [0x7f3b451f3546]
#20  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7f3b451ecfd8]
#21  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f3b4520b3af]
#22  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x1faba6) [0x7f3b451faba6]
#23  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7f3b452299df]
#24  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7f3b452291f7]
#25  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x23e3be) [0x7f3b4523e3be]
#26  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3546) [0x7f3b451f3546]
#27  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7f3b451ecfd8]
#28  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f3b4520b3af]
#29  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5704) [0x7f3b451d5704]
#30  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x1035dd) [0x7f3b451035dd]
#31  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f3b4520b3af]
#32  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5704) [0x7f3b451d5704]
#33  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallFunction+0x350) [0x7f3b451d5380]
#34  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/./libshiboken2.cpython-311-x86_64-linux-gnu.so.5.15(FinishSignatureInitialization+0x14a) [0x7f3b44ec1d6a]
#35  /tmp/.mount_FreeCAKbaCiN/usr/lib/python3.11/site-packages/PySide2/QtCore.cpython-311-x86_64-linux-gnu.so(PyInit_QtCore+0x27f4) [0x7f3b0ed572f4]
#36  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x2cb39c) [0x7f3b452cb39c]
#37  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x201e2b) [0x7f3b45201e2b]
#38  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7f3b451ecfd8]
#39  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f3b4520b3af]
#40  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x1faba6) [0x7f3b451faba6]
#41  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7f3b452299df]
#42  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7f3b452291f7]
#43  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x23e3be) [0x7f3b4523e3be]
#44  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3546) [0x7f3b451f3546]
#45  /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x82f8) [0x7f3b451ecfd8]
[... cut ...]
#106  0x7f3b45b9c56d in Gui::StartupPostProcess::execute() from /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libFreeCADGui.so+0x6d
#107  0x7f3b45ac8ac2 in Gui::Application::runApplication() from /tmp/.mount_FreeCAKbaCiN/usr/bin/../lib/libFreeCADGui.so+0x4e2
#108  /tmp/.mount_FreeCAKbaCiN/usr/bin/freecad(+0x7877) [0x55d5b0afe877]
#109  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f3b42c29d90]
#110  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f3b42c29e40]
#111  /tmp/.mount_FreeCAKbaCiN/usr/bin/freecad(+0x7bea) [0x55d5b0afebea]
```

luzpaz, 2026-10-08:
ping @plundell