FreeCAD/FreeCAD #17311: Assembly: the little coordinate systems to move parts disappear

peter-ch, 2024-10-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm running RC2 on Windows 10. The little coordinate systems of all objects in the assembly just disappeared and I can't bring them back, so I can't move anything. Only right-click on the tree view and transform works, but why are they gone? This cripples my workflow, I can't basically continue when it happens. The last time this happened I created an object with PartDesign and added it to the assembly. Is it something to do with PartDesign? 

I saved my project and it seems the problem is saved along with the file because when I open it I still can't move anything. 

[image]

### Full version info

```shell
FreeCAD 1.0 RC2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-10-17:
try double clicking your assembly object to activate it

peter-ch, 2024-10-17:
Thanks, that worked. 

I tried before right-click and set "Active Object" but that didn't work.

adrianinsaval, 2024-10-17:
> Thanks, that worked.
> 
> I tried before right-click and set "Active Object" but that didn't work.

@PaddleStroke shouldn't this work too?

maxwxyz, 2024-10-17:
works too. In the screenshot it is not activated. Closing this issue.

closed 2024-10-17 by maxwxyz