FreeCAD/FreeCAD #17313: Scale placement missing

VinceBa, 2024-10-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When Clone (draft wb) feature keep the parent placement, Scale (part wb) forget it.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 (Git)
Build type: Release
Branch: main
Hash: e478192aa5393de9b7df190341a043f79ec6ac49
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * A2plus 0.4.68
  * Assembly3 0.12.2
  * Assembly4 0.50.15
  * Behave-Dark-Colors 0.1.1
  * BOLTSFC 2022.11.5
  * CfdOF 1.27.10
  * CurvedShapes 1.0.12
  * Curves 0.6.50
  * dodo 1.0.1
  * Dracula 0.0.5
  * ExplodedAssembly
  * exporttopovray-master
  * ExtremeProDark 2.5.3
  * fcgear 0.3.38
  * frameforge-main 0.1.0
  * freecad.gears 1.2.0
  * freecad.trails
  * Geomatics
  * Help 1.0.3
  * kicadStepUpMod 11.3.9
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * MbDyn
  * MOOC 2022.4.21
  * Nikra-DAP-main
  * offline-documentation 1.0.0-alpha
  * OpticsWorkbench 1.0.24
  * parts_library
  * Plot 2022.4.17
  * POV-Ray-Rendering
  * ProDarkThemePreferencePack 1.0.0
  * Pyramids-and-Polyhedrons
  * QuickMeasure 2022.10.28
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Trails 2022.1.0
  * Woodworking-master 0.20.29177
  * workfeature-macro
```

### Subproject(s) affected?

None

### Anything else?

[Scale_Issue.zip](https://github.com/user-attachments/files/17421572/Scale_Issue.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2025-01-03:
Confirmed in the already mentioned topic, and also in this additional one:
https://forum.freecad.org/viewtopic.php?p=801329#p800161

luzpaz, 2025-05-17:
This is still relevant, correct ?