FreeCAD/FreeCAD #17314: Constraints using unicode names can be created and referenced, but suggested incorrectly

flukas, 2024-10-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Reproduction steps: 
- empty document, create sketch with one circle, dimension it and set name of dimension to `aščřž`, close sketch
- 
- create new sketch, add a circle, dimension it with formula expression to `Sketch.Constraints.ščžř` => this works

Expected:
- correct constraint name that can be both saved and referenced is offered

Observed problem
- in formula field type `Sketch.Constraints.a` => `aÅ¡ÄÅÅ¾` is suggested
- this cannot be used (Result: Failed to parse expression)

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Czech/Czech Republic (cs_CZ)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-08-29:
@flukas can you provide an example file and possible screencast ?

PaddleStroke, 2026-09-07:
This issue was fixed by https://github.com/FreeCAD/FreeCAD/pull/29875

luzpaz, 2026-09-07:
Closed by #29875

closed 2026-09-07 by luzpaz