FreeCAD/FreeCAD #17318: Switching Draw-Styles causes several operations to not be applied

Samu152, 2024-10-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you

1. Create sketches in PartDesign
2. Switch the Draw-Style (for example to wireframe)
3. Perform an operation with the sketches (for example a loft) 
4. Switch the Draw-Style back to original

Then the contour of the resulting shape will be visible, but not applied to the model. See images below:

[image]
[image]
[image]

Tested with:
- Additive Loft
- Subtractive Loft
- Additive Pipe
- Subtractive Helix (Additive Helix works as expected)
- Pad works as expected
- Revolution works as expected
- Hole works as expected
- Pocket works as expected

This only happens when switching to either: Points, Wireframe or Hidden Lines

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Austria (de_AT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

PartDesign

### Anything else?

https://forum.freecad.org/viewtopic.php?t=91431

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Samu152, 2024-10-18:
Just noticed a pattern. This seems to happen only with operations that don't apply the preview directly to the model. For example AdditiveLoft shows the preview as this yellow color you can see on the image. But a pad for example doesn't do that. It directly applies the changes to the model and is therefore unaffected

maxwxyz, 2024-10-19:
@kadet1090 maybe that is something you've came across in https://github.com/FreeCAD/FreeCAD/pull/17249 ?

luzpaz, 2024-10-19:
Bug confirmed in forum thread

kadet1090, 2024-10-20:
> @kadet1090 maybe that is something you've came across in #17249 ?

No but I'll take a look within few days

kadet1090, 2024-10-24:
Yes it is totally caused by how preview is done currently. #17249 does not have the same issue present.

closed 2025-08-07 by chennes