FreeCAD/FreeCAD #17319: Change of pad length is not updated in 3D view

Panda631, 2024-10-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a pad 100mm length

[image]

Double click the Pad and change the length in the task panel to e.g. 50mm.
==> Result: nothing happens.

Make the Pad visible in tree view.
The pad itself has the correct length.

[image]

changing the pad length in the data window works as expected.

[image]

[PadLength.zip](https://github.com/user-attachments/files/17430921/PadLength.zip)

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * EM 2.1.1
  * fasteners 0.5.27
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * Pyramids-and-Polyhedrons
  * Render 2024.8.4
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-10-18:
I can reproduce this -- the Pad options panel change is not "taking". The measurement does change in the properties. Forcing a recompute doesn't help.

Making the change in the properties will immediately be reflected in the model.

If I remove the chamfer and the thickness, the problem goes away.

Experimenting -- deleting everything except the sketch, re-extruding, chamfering a corner -- it seems that the problem appears after the chamfer is added.

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * Manipulator 1.5.7
  * fasteners 0.5.29
  * lattice2 1.0.0
  * CurvedShapes 1.0.12
  * Curves 0.6.50
```

FlachyJoe, 2024-10-18:
Fixed by #17248 please update.

luzpaz, 2024-10-18:
Committed into aa78eb351ee5594e20dcec80dd72a2bc5297cc57

Would be nice to have a way to know what revision this commit was in, any idea @FlachyJoe ?

FlachyJoe, 2024-10-18:
@luzpaz 
```
$ git rev-list --count aa78eb351ee5594e20dcec80dd72a2bc5297cc57
38983
```

luzpaz, 2024-10-18:
Closing as a duplicate of https://github.com/FreeCAD/FreeCAD/issues/17210