FreeCAD/FreeCAD #17320: Font file problem

JackNoble1963, 2024-10-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I try to do fillet and get first error message that I need to active cvurrent shape. I double click the part, I get this error message: 14:02:29  DirectWrite: CreateFontFaceFromHDC() failed (Ilmaisee virhettä syötetiedostossa, kuten fonttitiedostossa.) for QFontDef(Family="8514oem", pointsize=10, pixelsize=20, styleHint=5, weight=75, stretch=100, hintingPreference=0) LOGFONT("8514oem", lfWidth=0, lfHeight=-20) dpi=120

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Finnish/Finland (fi_FI)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-18:
Can you test the weekly version, there was a fix made in #17275 that may address this.  
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

maxwxyz, 2024-11-01:
@JackNoble1963 please test in the latest weekly. If there is still an issue comment and I'll reopen.

closed 2024-11-01 by maxwxyz