FreeCAD/FreeCAD #17322: Persistent Section cut: assigning of colour not possible

Panda631, 2024-10-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Assigning a color is not reflected in the model

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * EM 2.1.1
  * fasteners 0.5.27
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * Pyramids-and-Polyhedrons
  * Render 2024.8.4
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-10-18:
I can confirm this. I guess it's caused by the changes in Color per face.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

luzpaz, 2024-10-18:
Was this issue searched for prior to reporting? https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aopen+is%3Aissue+label%3A%22Topic%3A+Color%22

Panda631, 2024-10-18:
> Was this issue searched for prior to reporting? https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aopen+is%3Aissue+label%3A%22Topic%3A+Color%22

Sorry, I couldn't find this in your linked issues.
(Maybe I'm not a native english speaker.)

luzpaz, 2024-10-18:
Labeling blocker to get discussion and consensus if should be fixed in 1.0

maxwxyz, 2024-10-19:
@davesrocketshop FYI
Duplicate  of https://github.com/FreeCAD/FreeCAD/issues/15642

closed 2024-10-19 by maxwxyz