FreeCAD/FreeCAD #17323: Feature: Include Layers in Draft_AddToGroup tool

semhustej, 2024-10-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, Draft_AddToGroup does not support Layers. When moving elements to layers in mid sized-projects in model tree, it becomes quite tedious. Having Draft_AddToGroup support layers will speed up the workflow quite extensively.

**Proposal:**
Add Layer support to Draft_AddToGroup

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Draft

### Anything else?

To consider: possible creation of a new tool Draft_AddToLayer, next to Draft_AddToGroup.

I am not in favor of this. I think having just one tool with both Group and Layer included might be easier to manage. Layers and groups are also merged in the Draft_Autogroup tool.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-11-27:
Objects can be in a layer AND a group. This raises some questions.

Case 1: We have an object in group G, and we move it to layer L. What should happen?

Case 2: We have an object in group G and in layer L. What should selecting the "Ungroup" option then do?

semhustej, 2024-11-27:
Case 1: When moving such an object in the project tree, it is placed in both the group and the layer. So I think it makes sense to use the same behavior even for Draft_AddToGroup. _(Case 1 - object stays in group G and is also placed in layer L)_

Case 2: This is more tricky. Ideally, FreeCAD would detect ungrouping such an objects and pop up another window asking whether to ungroup from Layer (name) or Group (name). But I guess this is not that easy to code and than we will also have the problem of ungrouping several selected objects, which might be in different groups/layers. 
So for the sake of simplicity I would ungroup the object from both the group and the layer. This would also follow the name of the command 'Ungroup' - the object will not be part of any group/layer after it is processed. _(Case 2 - object is taken out of group G and out of layer L)_

This is my opinion. Bear in mind, that I have no idea why it was decided, that object can be part of a layer and also a group.

PR #19427 (2025-02-07): Draft: implement Draft_AddToLayer command
Fixes #17323.

* Instead of extending Draft_AddToGroup a separate, but similar, command has been created.
* The three "Add to" tools have been grouped in the menu and the toolbar.
* The Draft_AddToGroup.svg icon is provisional. Ideally all "Add to" tools should have related icons (it would then be nice to keep the trowel image in some way for the construction group). Draft_AddNamedGroup.svg and Draft_SelectGroup.svg probably also need to be updated in that context.

[image]
commit be0f3f9c (2025-02-07): Draft: implement Draft_AddToLayer command

Fixes #17323.

* Instead of extending Draft_AddToGroup a separate, but similar, command has been created.
* The three "Add to" tools have been grouped in the menu and the toolbar.
* The Draft_AddToGroup icon is provisional. Ideally the "Add to" tools should have related icons (it would then be nice to keep the trowel image in some way for the construction group). Draft_AddNamedGroup.svg and Draft_SelectGroup.svg probably also need to be updated in that context.
closed 2025-02-07 by Roy-043