FreeCAD/FreeCAD #17332: loguru package not support

xbkaishui, 2024-10-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi: 

There is a problem when use loguru in Mod, when I import loguru, it throws a error, details info is : 

> from loguru import logger
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "E:\programs\FreeCAD_weekly-builds-38429-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\programs\FreeCAD_weekly-builds-38429-conda-Windows-x86_64-py311\bin\Lib\site-packages\loguru\__init__.py", line 31, in <module>
    logger.add(_sys.stderr)
  File "E:\programs\FreeCAD_weekly-builds-38429-conda-Windows-x86_64-py311\bin\Lib\site-packages\loguru\_logger.py", line 755, in add
    if isinstance(sink, (str, PathLike)):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen abc>", line 119, in __instancecheck__
AttributeError: __class__

### Full version info

```shell
freecad 0.22.dev
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-19:
Please check out how to [report issues](https://github.com/FreeCAD/FreeCAD?tab=readme-ov-file#reporting-issues) which could help us diagnose and fix it.

- add full [About](https://wiki.freecad.org/About) info to the issue (`Help > About FreeCAD > Copy to clipboard`).
- share an example file with the issue present. You can upload a zipped file to the original issue.
- add steps to reproduce the issue

luzpaz, 2024-10-19:
* Please post your full [About](https://wiki.freecad.org/About) info.
* It looks like you're running an older version of the weekly build please update to the latest one (1.1dev) or RC2
* What is logguru? Is it a 3rd-party python library?

xbkaishui, 2024-10-19:
Sorry, the full About info is:

> OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38429 (Git)
Build type: Release
Branch: main
Hash: 362cf1c80edb89957d81c0dbe6e50ce35daaa6db
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Chinese/China (zh_CN)
Installed mods: 
  * CurvedShapes 1.0.9
  * Curves 0.6.42
  * Defeaturing 1.2.2
  * freecad_robot_python.zip
  * Manipulator 1.5.7
  * RobotDescriptor
  * robot_workbench
  * workfeature
  * workfeature-macro

and loguru is a python logger library.

xbkaishui, 2024-10-19:
The reproduce step is: 

1. install loguru using python,  {FREECAD_INSTALL_DIR}\bin\python.exe -m pip install loguru
2. in freecad python console window,  type  import loguru, then the error shows

It's wired when I open freecadcmd.exe and import loguru, there is no error

please help

luzpaz, 2024-10-19:
@xbkaishui 
1st, you're using an very obsolete dev build of FreeCAD. If you're going to run dev builds you need to keep up with the latest one so that you report accurate bugs. Get the latest weekly dev build at https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

2nd, https://github.com/Delgan/loguru is a 3rd party python library that FreeCAD isn't designed 'out of the box' to use. So just downloading any package via pip and running it in FreeCAD does not mean it will work. 

3rd, what are you trying to achieve? What's your end goal?

4th, if you're interested in making FreeCAD and loguru compatible, then you'll probably need to open a forum.freecad.org thread and discuss with the community strategies to do so. And then most likely someone needs to submit changes to the  projects (freecad and loguru) to make it possible.

luzpaz, 2024-10-19:
Closing based on point No. 4 above.

closed 2024-10-19 by luzpaz

xbkaishui, 2024-10-20:
Hello, got it. I will open a thread