FreeCAD/FreeCAD #17333: Weird message pops up, QtWaylandClient

GAZ082, 2024-10-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When i fire off RC2 a number of pops up start repeatedly showing for a fraction of a second. Checking the notification area I get this:

[image]

https://github.com/user-attachments/assets/5ce0deec-33ad-4e40-b4a6-30060269ebac

This also happens when i create a new document, until I click somewhere.

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: makepkg
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Installed mods: 
  * OpenTheme 2024.9.1 (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-10-19:
This seems to be an upstream Qt issue; Qt are still not fully functional on Wayland.

However, these pop ups shouldn't be shown to the user, but hidden (for instance only shown in terminal).

luzpaz, 2024-10-19:
Looks like a dupe of #16687

closed 2024-10-19 by luzpaz