FreeCAD/FreeCAD #17335: FC 1.1.0dev.38977 : Hole cut not exported into STL when hole is "real tip" of object

DerMouse, 2024-10-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When drilling holes in PartDesign, if

1. hole has a cut type (counterbore, countersink, etc) and
2. hole is the LAST item and tip of the object

then the cut type will not be exported into the STL file (the hole will be).

In FreeCad :
[image]

STL file under PrusaSlicer
[image]

The FCStd file :
[Hole-1.zip](https://github.com/user-attachments/files/17444457/Hole-1.zip)

Exported STL file : 
[Hole-1-Body.zip](https://github.com/user-attachments/files/17444462/Hole-1-Body.zip)

If *anything* is created after the current tip, even if only an empty sketch that is then deleted, the cut type will be properly exported.

FCStd file, I added a sketch after the hole and then deleted the sketch :
[Hole-2.zip](https://github.com/user-attachments/files/17444458/Hole-2.zip)

Exported STL file : 
[Hole-2-Body.zip](https://github.com/user-attachments/files/17444464/Hole-2-Body.zip)

The problem will reappear if, later on, another hole with any cut type is the last element/true tip of the object.

It will also raise its ugly head if, in the Hole-2.FCStd file (properly exporting the cut type because of the "empty sketch" workaround), the cut type is changed (from countersink to counterbore for example). Then, the exported STL file will still show the countersink cut type.

The problem does not appear if the hole is only set as tip, but there are other objects (not exported) after it in the tree.

### Full version info

```shell
OS: Gentoo Linux (xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git) AppImage
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * freecad.gears 1.2.0
  * persistenttoolbars
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-10-19:
Hi, current dev is 38991 can you update and re-test?

DerMouse, 2024-10-19:
> Hi, current dev is 38991 can you update and re-test?

Same thing with 

OS: Gentoo Linux (xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git) AppImage
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * freecad.gears 1.2.0
  * persistenttoolbars

DerMouse, 2024-10-19:
> > Hi, current dev is 38991 can you update and re-test?
> 
> Same thing with

Well as a matter of fact, not exactly the same behaviour.

Opening the Hole-1.FCStd file under FC-38991 and exporting it to STL doesn't enhance anything, cut type still wrong. But 

1. if completely creating the file under FC-38991, or
2. updating the Hole-1 file under FC-38991 (saving it without change is not enough), _even_ if adding a new tip with a hole and some cut type,

problem does not appears anymore.

maxwxyz, 2024-10-19:
so I'll close this as fixed in latest dev

closed 2024-10-19 by maxwxyz