FreeCAD/FreeCAD #17336: Clipping plane: Transform dragger is cut / invisible

maxwxyz, 2024-10-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using clipping planes to see inside the model/assembly, the draggers/gizmos of the transform tool are cut too and/or iinvisible so it is not possible to rotate / move the part.

### Full version info

```shell
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.50
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-19:
@kadet1090 maybe you know a hack around as you've improved it already?

kadet1090, 2024-10-19:
In So3DAnnotation we could disable the clip plane so it will render fully.

maxwxyz, 2025-09-05:
@kadet1090 @captain0xff also applies to:
- new axis cross
- interactive draggers

[image]