FreeCAD/FreeCAD #17338: Part: Boolean Union / Fusion fills spaces where no part exists

TadPath, 2024-10-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On the Part workbench, doing a simple Boolean fusion between two valid solid shapes results in a union that has filled in regions where none of the shapes contain solid areas. See example where thread gaps are filled in after union of thread with solid cylinder.

[image]

This problem only occurred with transition to v.1.0 RC - it does not happen with v0.2.

Here is a link to the FreeCAD file in case you wish to confirm the problem:

[Thread_Union_Test_FreeCAD_File ](https://drive.google.com/file/d/13kxA1OhlJIx_qQGXIUhFUd49cRi0nXWD/view?usp=sharing)

Details: Part -> Check Geometry on the thread component shows no invalid components and the Sketch is fully constrained (also, this union worked perfectly well with older versions of FreeCAD).
No errors, warnings or messages in the Report View.

Comment: Please note. I am aware that there may well be many alternative or even 'better / more correct' ways of making a solid thread than fusing a cylinder with a sweep solid but this is how I do it (and have hundreds of these in my PUMA model files which all worked fine till RC1.0 so it would be impractical to go back and re-do them all in some alternative way).
The issue is one of the Boolean operation not working as expected rather than how to achieve the specific shape illustrated in the example given. Also, I am aware that the cylinder intersects the thread at the periphery - but this is deliberate - and worked fine in all previous versions of FreeCAD - as a Boolean Union of two solids would be expected to work. Thanks.

### Full version info

```shell
OS: Arch Linux (LXDE/LXDE)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: makepkg
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Installed mods: 
  * fasteners 0.5.29
  * OpticsWorkbench 1.0.24
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-10-19:
confirmed in latest main, OCC issue

AIRCAP, 2024-10-19:
not fixed by fuzzyness - or changing tolerances - this is a genuine OCCT bug it seems

AIRCAP, 2024-10-19:
workaround: -- with #17119 the cylinder can be set to exactly 17.82 mm radius (the inner diameter of the thread) without overlap. that prevents this issue. However in main this isn't possible since it freezes FreeCAD

AIRCAP, 2024-10-19:
@TadPath  understood. Yes, the issue is FreeCAD use a library for atomic shape operations such as fusing two parts. All the FreeCAD opeartion to fuse the cylinder with the threaded part does is call the underlying OCCT fuse function, which does all the math to compute a shape. This is buggy - in some versions at least. It's possible that when compiling FreeCAD against a newer version of that library this bug goes away, or not. We can't do much about it. (The maintainers of OCCT make it quite hard to submit fixes even if we could fix this)

Where we can do something is this:

> as soon as the two solids **touch** or ~~intersect~~, the Boolean fusion fails

the behaviour - and bug - is different. Where on intersection you get this weird effect where there are closed faces where they should not be - if they touch, you get even weirder behaviour, including sometimes crashes, freezes, or also broken geometry - but with different symptoms.  its two different bugs. --  this touching is the case with 17.82 with this specific part.

In your case doing that might not be an option, but in the general case, having two shapes that touch combined into one is what people would normally do, and only if that doesn't work resort to workarounds such as deliberate intersection between the parts.

Where we CAN do something about is the other bug - which happens when the parts are touching, because FreeCAD is at least partially at fault there. OCCT is supposed to be able to detect if two parts are just touching, and we are working on a fix for that.

So while this might be unsatisfying for your specific case and your work, an issue here is primarily treated under the aspect of "what can we learn from this and how can we make FreeCAD better" 

in this case the lesson learned is that making parts overlap to fix bug 1 - triggers bug 2 - which we have no control over. so we really need to fix bug 1

that's what I wanted to write in my original post, which was mostly meant as informative additional info for the triage group

TadPath, 2024-10-19:
OK @AIRCAP , I only wanted to make clear what the issue is - not criticising.

A Boolean Union (or fusion or whatever is the correct term) surely must include cases where the objects overlap and not just are either separate or just touching. Otherwise it would not be a proper logical operation - a bit like saying you can only do an intersection of two objects provided they don't overlap - not very useful.

Having said this, I just tried the 'Intersection' function of FreeCAD on this example file and it also fails - it results in nothing (when there clearly is some overlap between the solids) - so the both the fusion (union) and intersection Boolean operations fail in this case.

As before - please note that these problems only occur in the new RC 1.0 versions. The pre 1.0 versions of FreeCAD did the union / fusion and intersection flawlessly so I hope that this can be worked out in FreeCAD unless the OCCT lib has simulatenously / coincidentally developed a bug.

AIRCAP, 2024-10-19:
its possible that the earlier freecad version was compiled against an earlier version of OCCT -  can you provide the full version info for the 0.21.1 freecad with which the issue does not happen, including OCCT version?

AIRCAP, 2024-10-19:
I just tested with freecad 0.21.1 with OCCT 7.6.3 and that shows the same error

AIRCAP, 2024-10-19:
1.1dev with OCCT 7.3.0 also has the issue

AIRCAP, 2024-10-19:
[code]
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
 even 0.20.2 - OCC 7.6.3 - same issue

TadPath, 2024-10-20:
Hi @AIRCAP , I would like to do what you did and go back to where the trouble starts but I am having difficulty running an older version of FreeCAD on my Arch machine. When I try to run a 0.2 AppImage from the Git I get a GLX init arror thus:

[image]

How did you run the old versions? Do you compile from source?

TadPath, 2024-10-22:
Please note: The back version results obtained by @AIRCAP were done on the FreeCAD file I provided here which was edited and saved in FreeCAD v. 1.0RC. Therefore it is possible that v1.0RC has saved the information in a way that fails in older versions.
Once I have figured out how to load older versions of FreeCAD on my machine (overcome this GLX problem I posted about above) I can then test original files made with v0.2.x and provide the version info as to when this error first occurs.

stefano2734, 2026-09-12:
Is it version 0.2.x or 0.20.x with last good version? 
And which occt version is here related to the last good version?
I think it is important to see here what changes are here relevant for this problem.