FreeCAD/FreeCAD #17339: crash

sova32, 2024-10-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

freecad last version for linux crashes randomly! i dont know how to reproduse!
i was change sphere position and it crash, i was deket scatch and pad - crash. i can-not use isometric view!
with evry update it`s more bags. in near future it vill 100% unusable!

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: makepkg
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: Russian/Ukraine (ru_UA)
Installed mods: 
  * freecad.gears 1.2.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-19:
Please use RC2 (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2) and report back.

xtemp09, 2024-10-20:
> ```shell
> OS: Manjaro Linux (KDE/plasma)
> ```

> ```shell
> Qt 6.7.2
> ```

A weekly build is likely to work fine.

luzpaz, 2024-10-20:
Per repology manjaro stable hasn't received RC2 just yet
[image]

luzpaz, 2024-10-20:
Closing. Please notate this ticket if you can still reproduce issue after trying suggested solutions.

closed 2024-10-20 by luzpaz