FreeCAD/FreeCAD #17340: CLI: BoundBox derived Shapes have wrong dimensions when recomputed in CLI

AIRCAP, 2024-10-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This shape has a torus and a sort of magnet like rectangular shape around it. 
It's a Sketch based extrusion, and the Sketch has constraints on the <<Torus>> boundingbox

[image]

The Draft Label prints the volume of the "magnet" - 614.000 mm³

there is a macro that recomputes the document, then prints the text of this label.

When executed from FreeCAD with GUI this prints
`FreeCAD BBox_Derived_Sketch.FCStd print_volume.py 
FreeCAD 1.1.0, Libs: 1.1.0devR38991 +37 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

['614,0000 mm³']
`
and saves a file broken.FCStd which is just identical to BBox_Derived_Sketch.FCStd

However if you do that with FreeCADCli you get:

```
FreeCADCmd BBox_Derived_Sketch.FCStd print_volume.py 
FreeCAD 1.1.0, Libs: 1.1.0devR38991 +37 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Importing project files......
Postprocessing......				(100 %)		
['653,5483 mm³']
```
suspicious, wrong volume ... let's look at broken.FCStd

1st the label is not displayed instead theres an error
```
15:16:34  v0.19, dLabel, added view property 'ScaleMultiplier'
15:16:34  v0.19, dLabel, added view property 'AnnotationStyle'
15:16:34  pyException: Traceback (most recent call last):
  File "/mnt/slow1/src/FreeCAD/main/Mod/Draft/draftobjects/label.py", line 227, in onDocumentRestored
    super().onDocumentRestored(obj)
  File "/mnt/slow1/src/FreeCAD/main/Mod/Draft/draftobjects/draft_annotation.py", line 69, in onDocumentRestored
    self.add_missing_properties_0v19(obj, vobj)
  File "/mnt/slow1/src/FreeCAD/main/Mod/Draft/draftobjects/draft_annotation.py", line 81, in add_missing_properties_0v19
    vobj.Proxy.set_annotation_properties(vobj, vobj.PropertiesList)
<class 'AttributeError'>: 'NoneType' object has no attribute 'set_annotation_properties'
```
2nd and more importantly, the part has the wrong shape:

[image]

only after a complete recompute is the shape correct again

[image]

[bigmess.zip](https://github.com/user-attachments/files/17445359/bigmess.zip)

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (XFCE/xfce/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 +37 (Git)
Build type: Debug
Branch: fixBooleans,debugTolerance
Hash: c3068fee8082c905a9bf50e284cd8f9b23f0e19c
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

(also happens in unmodified main - and probably in all FreeCAD versons)
```

### Subproject(s) affected?

Core

### Anything else?

The reason is that BoundBox relies on the tessellation of the object. If the shape has not been tessellated, boundbox does not give correct results. Since the Sketch is boundbox derived, it will get the wrong dimensions, resulting in wrong shapes.

Now the issue is that parts are being tessellated based on ViewProvider configuration (tessellation accuracy is specified in the viewprovider properties) and this does not happen at all in CLI

This is a problem. Part Shape and properties in the App feature should never depend on viewprovider parameters, yet BoundBox does.

CLI is just an extreme case where the tesselation doesn't happen at all, however the exact dimensions of the boundbox can also depend on the tessellation accuracy used for display. This is highly undesirable and breaks FreeCADs assumptions that shape data does not depend on viewprovider information.

A part could have a different shape depending on the users tessellation preferences.

That's a architectural flaw.

I will open a second issue for the broken label, if there isn't one already.

Suggestion:

1. Quick fix: Have a Property in TopoShape that marks if tessellation has happened yet. If tessellation has not happened, any call to BoundBox should trigger tessellation, and set the flag to true. That tessellation should have a fixed size relative accuracy (not too large) and depend on the size of the part to limit computation time for large shapes.

2. Proper fix: Completely detach tessellation for display from tessellation for boundingbox computation. compute a fixed accuracy (or accuracy accoding to a data property)  tessellation as part of each shape ( at last if boundbox is used )  -- compute separate tessellation with user defined accuracy for display
OR
2. Proper fix: Move the tessellation settings (part and angular accuracy) from the view provider properties into data properties for all shapes - use this setting for both visual display and computation -- also trigger tessellation when boundbox is requested

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-10-21:
[BRepBndLib::AddOptimal()](https://dev.opencascade.org/doc/refman/html/class_b_rep_bnd_lib.html#aad43f924ff3d1ff0963005c4c6cdeb90) uses a more precise method to compute the BBox

AIRCAP, 2024-10-21:
So we'd need to use AddOptimal everywhere where BoundingBoxes are used for shape calculations - including the python code that provides the contents of `Shape.BoundBox` ?
What would be the impact on performance of doing that?

closed 2025-08-30 by github-actions[bot]