FreeCAD/FreeCAD #17342: Draft: CLI: Labels broken in documents stored in headless

AIRCAP, 2024-10-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When a FCStd is stored headless mode without GUI (for example from FreeCADCmd) then Draft labels get broken - permanently

When opening the file in FreeCAD and recomputing the following error happens:

```
15:16:34  v0.19, dLabel, added view property 'ScaleMultiplier'
15:16:34  v0.19, dLabel, added view property 'AnnotationStyle'
15:16:34  pyException: Traceback (most recent call last):
  File "/mnt/slow1/src/FreeCAD/main/Mod/Draft/draftobjects/label.py", line 227, in onDocumentRestored
    super().onDocumentRestored(obj)
  File "/mnt/slow1/src/FreeCAD/main/Mod/Draft/draftobjects/draft_annotation.py", line 69, in onDocumentRestored
    self.add_missing_properties_0v19(obj, vobj)
  File "/mnt/slow1/src/FreeCAD/main/Mod/Draft/draftobjects/draft_annotation.py", line 81, in add_missing_properties_0v19
    vobj.Proxy.set_annotation_properties(vobj, vobj.PropertiesList)
<class 'AttributeError'>: 'NoneType' object has no attribute 'set_annotation_properties'
```
It seems the label relies on properties stored with the viewprovider that can't be re-generated. These should not be in the viewprovider in the first place but in the Data object.

for reproduction details see #17340

[bigmess.zip](https://github.com/user-attachments/files/17445430/bigmess.zip)

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (XFCE/xfce/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 +37 (Git)
Build type: Debug
Branch: fixBooleans,debugTolerance
Hash: c3068fee8082c905a9bf50e284cd8f9b23f0e19c
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.12
  * CfdOF 1.27.10
  * Curves 0.6.48
  * AirPlaneDesign 0.4.1
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-19:
@Roy-043 FYI

yorikvanhavre, 2024-10-21:
Indeed the App object should not mess with Gui stuff, however since the view provider has no onRestore callback, this is the only way to trigger something in the Gui space... The above PR adds a deeper test to make sure we have a functional view provider. @Roy-043 could you have a look if that solution seems good to you?

PR #17379 (2024-12-06): Draft: better check for Gui object
fixes #17342
commit 68be845a (2024-12-06): Draft: better check for Gui object - fixes #17342
closed 2024-12-06 by Roy-043