FreeCAD/FreeCAD #17346: Draft: Trimex on B-spline or Bezier curve produces errors

semhustej, 2024-10-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Priority: low

When trying to Draft_Trimex B-Spline or Beziere curve, FreeCAD outputs error. Steps to reproduce:

Steps to reproduce:

- Open file 
[b-spline-bezier.zip](https://github.com/user-attachments/files/17446758/b-spline-bezier.zip)

- Select B-spline or Beziere curve
- Start Draft_Trimex
- Type in distance and hit enter
- Error shows up

See behavior in the video:

https://github.com/user-attachments/assets/317f54d2-d8e5-495e-a32e-25045ab9cb09

**Proposed change:**
Post a message: `Trimex is not supported on this type of object.`

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-20:
Can confirm. @Roy-043 I guess B-Splines are not supported yet?

yorikvanhavre, 2024-10-22:
Indeed b-curves are not supported by that tool yet. Maybe we could just add a warning for now?

semhustej, 2024-10-22:
> **Proposed change:**
> Post a message: `Trimex is not supported on this type of object.`

Yes, that is proposed solution for both  B-splines and Beziere curves too:

Proposed change:
Post a message: `Trimex is not supported on this type of object.`

PR #17456 (2024-10-28): Draft: Warn user if using trimex on unsupported curve type
Fixes #17346
commit dc230a8c (2024-10-28): Draft: Warn user if using trimex on unsupported curve type - fixes #17346
closed 2024-10-28 by yorikvanhavre