FreeCAD/FreeCAD #17352: 3DConnexion support aint alright

APOS80, 2024-10-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using a spacemouse in FreeCAD is not helping as it doesnt seem to be implemented well.

It would be nice to be able to fly fpv around an object, its the natural way to us it in other software.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Swedish/Sweden (sv_SE)
Installed mods: 
  * BIM 2021.12.0
  * QuickMeasure 2022.10.28
  * Trails 2022.1.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-20:
please try it on the dev build: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds or the 1.0 release candidates

luzpaz, 2024-10-20:
Closing. If issues persist, please specify what they are.

closed 2024-10-20 by luzpaz

APOS80, 2024-10-25:
Flying can be set tp fpv now but speed is a problem.

Even if i set to slowest its a bit fast, I recomend to set it lower in code,

luzpaz, 2024-10-25:
@APOS80 can you open a new ticket for that ? please add your full [About](https://wiki.freecad.org/About) info to it