FreeCAD/FreeCAD #17353: BIM_GenericTools error when text is typed into the search field

semhustej, 2024-10-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Start BIM_GenericTools (3D/BIM -> Generic 3D Tools -> Objects library)
2. Type text into the search field
3. Error shows up:

```
12:19:50  Traceback (most recent call last):
12:19:50    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-38991-conda-Windows-x86_64-py311\Mod\BIM\bimcommands\BimLibrary.py", line 377, in onSearch
12:19:50      self.setSearchModel(text)
12:19:50    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-38991-conda-Windows-x86_64-py311\Mod\BIM\bimcommands\BimLibrary.py", line 392, in setSearchModel
12:19:50      for dp, dn, fn in res:
12:19:50          ^^^^^^^^^^
12:19:50  ValueError: too many values to unpack (expected 3)
12:19:50  Traceback (most recent call last):
12:19:50    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-38991-conda-Windows-x86_64-py311\Mod\BIM\bimcommands\BimLibrary.py", line 377, in onSearch
12:19:50      self.setSearchModel(text)
12:19:50    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-38991-conda-Windows-x86_64-py311\Mod\BIM\bimcommands\BimLibrary.py", line 392, in setSearchModel
12:19:50      for dp, dn, fn in res:
12:19:50          ^^^^^^^^^^
12:19:50  ValueError: too many values to unpack (expected 3)
12:19:50  Traceback (most recent call last):
12:19:50    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-38991-conda-Windows-x86_64-py311\Mod\BIM\bimcommands\BimLibrary.py", line 377, in onSearch
12:19:50      self.setSearchModel(text)
12:19:50    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-38991-conda-Windows-x86_64-py311\Mod\BIM\bimcommands\BimLibrary.py", line 392, in setSearchModel
12:19:50      for dp, dn, fn in res:
12:19:50          ^^^^^^^^^^
12:19:50  ValueError: too many values to unpack (expected 3)

```

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

BIM

### Anything else?

The error was confirmed by another user in this forum topic: https://forum.freecad.org/viewtopic.php?t=91517

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-20:
@yorikvanhavre FYI

yorikvanhavre, 2024-10-22:
Confirmed. Working on a fix...

commit 1c6e5b96 (2024-10-28): BIM: Fixed online search mode - fixes #17353